CameraKit/camerakit-android #2: Problems on LG G3

balzss, 2017-02-28:
Hi,
Thanks for putting this project together. Previously I had to deal with the Camera2 API and it's just a nightmare so I highly appreciate every project like this. However the demo app is not working perfectly on my LG G3 (D855). When I press the "camera turn" button the app closes and give a message that "Unfortunately, CameraKit has stopped." The other thing is the flashlight button: it does not turn on the flashlight. I know these statements are very generic but I don't know what else I could provide. I hope this helps a little.

muetzenflo, 2017-02-28:
i can confirm this (also on LG G3). I cloned the repo to see what's going on, but in debug mode everything works as expected. This seems to be a proguard issue => crash log in dev console

ThomasKomarnicki, 2017-02-28:
I've seen the same issue on my Nexus 5 (4.4) and Note 5 (6.0). It looks like the culprit is Camera1:261:
 `AspectRatio targetRatio = aspectRatios.size() > 0 ? aspectRatios.last() : null;` 
which throws:
```Caused by: java.util.NoSuchElementException
at java.util.TreeMap.lastKey(TreeMap.java:630)
at java.util.TreeSet.last(TreeSet.java:272)
at com.flurgle.camerakit.Camera1.getPreviewResolution(Camera1.java:261)
``` 

I'm not sure how `aspectRatios.last()` could throw an exception if the TreeSet has entries.

dwillmc, 2017-02-28:
@ThomasKomarnicki you see this on the app downloaded from Google Play right? That exception should be resolved in the library.

ThomasKomarnicki, 2017-02-28:
yes it's only happening in the demo downloaded from Google Play, I cloned the repo and ran the most up to date version and it is no longer happening

ThomasKomarnicki, 2017-02-28:
@dwillmc ah I found the commit where you fixed it, should be fine then..

dwillmc, 2017-02-28:
If you're curious, it happens due to me wrongly assuming that there will always be matching aspect ratios available when comparing the set of available preview sizes and available picture capture sizes.

dwillmc, 2017-03-10:
An update was just pushed which should rollout in the next few hours and fix this problem.
Please re-open if you get the update and it still doesn't work!

closed 2017-03-10 by dwillmc