CameraKit/camerakit-android #5: Crash when switching to Front Camera

aftabsikander, 2017-03-01:
Play store Application is getting crashed when switching Front Face Camera from option

# Device Information
```
Brand: Google
DeviceModel: Nexus 5
Product: hammerhead
Manufacturer: LGE
Android Version: 6.0.1
```

**How to reproduce this issue.**

- Download Application From Google Play Store

- Open Application give permissions and switch camera to **Front Face**, As soon as you click button application is crashed

baurine, 2017-03-02:
@aftabsikander it's true, it also crashed in my phone, try the latest code in the master branch, that's ok on my phone. this bug should be fixed in the commit: [fix flipping camera](https://github.com/flurgle/CameraKit-Android/commit/0a2a94245c31b628da7c1b1d9153b8db497e0e12)

sohayb, 2017-03-09:
@aftabsikander did the issue get resolved as @baurine stated? if yes, can you please close the issue?

aftabsikander, 2017-03-09:
@sohayb I would test this by day end, Will close this once I check it out

dwillmc, 2017-03-10:
An update was just pushed which should rollout in the next few hours and fix this problem.
Please re-open if you get the update and it still doesn't work!

closed 2017-03-10 by dwillmc