CameraKit/camerakit-android #6: Crashed on multiple devices when opening an application

DanishAmjad12, 2017-03-01:
Application crashed when opening the application & give the permission.

# Devices Information
```
Brand: Samsung
Device Model: Galaxy S6
Product: SM-G290V
Manufacturer: Samsung
Android Version: 6.0.1

Brand: Samsung
Device Model: Note 5
Product: SM-N920A
Manufacturer: Samsung
Android Version: 5.1.1

Brand: Motorola
Device Model: Droid turbo
Manufacturer: Motorola
Android Version: 6.0.1
```

**Crash Logs**

```
E/AndroidRuntime: FATAL EXCEPTION: main
Process: com.flurgle.camerakit.demo, PID: 14283
java.lang.RuntimeException: Unable to resume activity {com.flurgle.camerakit.demo/com.flurgle.camerakit.demo.MainActivity}: java.util.NoSuchElementException
at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4156)
at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4250)
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3361)
at android.app.ActivityThread.access$1100(ActivityThread.java:222)
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1795)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:158)
at android.app.ActivityThread.main(ActivityThread.java:7229)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
Caused by: java.util.NoSuchElementException
at java.util.TreeMap.lastKey(TreeMap.java:630)
at java.util.TreeSet.last(TreeSet.java:272)
at com.flurgle.camerakit.Camera1.getPreviewResolution(Camera1.java:261)
at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:333)
at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:292)
at com.flurgle.camerakit.Camera1.start(Camera1.java:74)
at com.flurgle.camerakit.CameraView.start(CameraView.java:151)
at com.flurgle.camerakit.demo.MainActivity.onResume(MainActivity.java:95)
at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1286)
at android.app.Activity.performResume(Activity.java:6987)
at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4145)
at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4250) 
at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3361) 
at android.app.ActivityThread.access$1100(ActivityThread.java:222) 
at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1795) 
at android.os.Handler.dispatchMessage(Handler.java:102) 
at android.os.Looper.loop(Looper.java:158) 
at android.app.ActivityThread.main(ActivityThread.java:7229) 
at java.lang.reflect.Method.invoke(Native Method) 
```

**How to reproduce this issue.**

- Download Application From Google Play Store

- Open Application give permissions then crashed occurred.

weklund, 2017-03-08:
+1 on Application crash

Brand: Huawei 
Device Model: Nexus 6P
Product: H1511
Android Version: 7.1.1

dwillmc, 2017-03-10:
An update was just pushed which should rollout in the next few hours and fix this problem.
Please re-open if you get the update and it still doesn't work!

closed 2017-03-10 by dwillmc