CameraKit/camerakit-android #12: NPE when resuming app

Crysis21, 2017-03-02:
device Galaxy S4 - 5.0.1

app was behind lockscreen. When I unlocked the phone, NPE occurred.
`
03-02 19:39:46.276 23113-23113/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.anaface.anaface, PID: 23113
                                                   java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getHeight()' on a null object reference
                                                       at com.flurgle.camerakit.CameraView.onMeasure(CameraView.java:135)
                                                       at android.view.View.measure(View.java:18580)
                                                       at android.widget.RelativeLayout.measureChildHorizontal(RelativeLayout.java:728)
                                                       at android.widget.RelativeLayout.onMeasure(RelativeLayout.java:464)
                                                       at android.view.View.measure(View.java:18580)
                                                       at android.support.v4.view.ViewPager.onMeasure(ViewPager.java:1658)
`

android-dataticket, 2017-03-06:
i add the photo to an adapter when its taken, when i click on the photo which opens it to view with gallery, i get this same NPE re: getHeight()

android-dataticket, 2017-03-06:
this crashing only when i click on the image in the adapter while in landscape, and the orientation changes when the image is opened.

if i click on the image while in portrait mode, it seems to not crash as the screen doesnt have to change orientation

dwillmc, 2017-03-10:
An update was just pushed which should rollout in the next few hours and fix this problem.
Please re-open if you get the update and it still doesn't work!

closed 2017-03-10 by dwillmc