CameraKit/camerakit-android #8: Tap to focus is not working.

Amirulhos, 2017-03-02:
I have put "app:ckFocus="tap"" in xml, but still tap to focus is not working, if possible please check and fix the issue.

lawonga, 2017-03-03:
Neither does pinch to zoom

caphson, 2017-03-04:
Confirmed, neither work either via XML or programatically.

Library: com.flurgle:camerakit:0.9.13
API: 24

calebdre, 2017-03-08:
[Here could be a solution](http://stackoverflow.com/a/28575143/2229572) if someone is willing to add it. I'll try my hand at it over the next week

calebdre, 2017-03-08:
[It doesn't look like `focus_tap` is implemented](https://github.com/flurgle/CameraKit-Android/blob/master/camerakit/src/main/api16/com/flurgle/camerakit/Camera1.java#L150)

I'm guess it's because you'd need to specify a focus area and manually invoke `camera.autofocus(paramters)` like in the SO example i posted. There would have to be a touch listener exposed in there somewhere though

closed 2017-04-03 by dwillmc