CameraKit/camerakit-android #15: crash when selecting photo uri in landscape

android-dataticket, 2017-03-06:
taking a picture, then clicking on the picture that was saved to view it as uri - this will cause NPE if the device is being held in landscape mode.  does not crash if the device is being held in portrait.. also we need a way to set orientation and be sure the photo always snaps at that rotation, please keep up the good work i'm relying on this lib now because i like it and it shows more promise than cameraview

java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getHeight()' on a null object reference
                                                                           at com.flurgle.camerakit.CameraView.onMeasure(CameraView.java:136)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1464)
                                                                           at android.widget.LinearLayout.measureVertical(LinearLayout.java:747)
                                                                           at android.widget.LinearLayout.onMeasure(LinearLayout.java:629)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1464)
                                                                           at android.widget.LinearLayout.measureHorizontal(LinearLayout.java:1111)
                                                                           at android.widget.LinearLayout.onMeasure(LinearLayout.java:631)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.support.v4.view.ViewPager.onMeasure(ViewPager.java:1658)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.LinearLayout.measureChildBeforeLayout(LinearLayout.java:1464)
                                                                           at android.widget.LinearLayout.measureVertical(LinearLayout.java:747)
                                                                           at android.widget.LinearLayout.onMeasure(LinearLayout.java:629)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.widget.RelativeLayout.measureChildHorizontal(RelativeLayout.java:716)
                                                                           at android.widget.RelativeLayout.onMeasure(RelativeLayout.java:462)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at android.support.v7.widget.ContentFrameLayout.onMeasure(ContentFrameLayout.java:139)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewGroup.measureChildWithMargins(ViewGroup.java:6328)
                                                                           at android.widget.FrameLayout.onMeasure(FrameLayout.java:194)
                                                                           at com.android.internal.policy.PhoneWindow$DecorView.onMeasure(PhoneWindow.java:3143)
                                                                           at android.view.View.measure(View.java:20171)
                                                                           at android.view.ViewRootImpl.performMeasure(ViewRootImpl.java:2644)
                                                                           at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2350)
                                                                           at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1487)
                                                                           at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7450)
                                                                           at android.view.Choreographer$CallbackRecord.run(Choreographer.java:920)
                                                                           at android.view.Choreographer.doCallbacks(Choreographer.java:695)
                                                                           at android.view.Choreographer.doFrame(Choreographer.java:631)
                                                                           at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:906)
                                                                           at android.os.Handler.handleCallback(Handler.java:739)
                                                                           at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                           at android.os.Looper.loop(Looper.java:158)
                                                                           at android.app.ActivityThread.main(ActivityThread.java:7237)
                                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
                                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)

android-dataticket, 2017-03-08:
this fixes the crash, see changes I made in bold: 

    @Override
    protected void onMeasure(int widthMeasureSpec, int heightMeasureSpec) {
        if (mAdjustViewBounds) {
            Size previewSize = getPreviewSize();
            **if(previewSize != null) {**
                if (getLayoutParams().width == LayoutParams.WRAP_CONTENT) {
                    int height = MeasureSpec.getSize(heightMeasureSpec);
                    float ratio = (float) height / (float) previewSize.getWidth();
                    int width = (int) (previewSize.getHeight() * ratio);
                    super.onMeasure(
                            MeasureSpec.makeMeasureSpec(width, MeasureSpec.EXACTLY),
                            heightMeasureSpec
                    );
                    return;
                } else if (getLayoutParams().height == LayoutParams.WRAP_CONTENT) {
                    int width = MeasureSpec.getSize(widthMeasureSpec);
                    float ratio = (float) width / (float) previewSize.getHeight();
                    int height = (int) (previewSize.getWidth() * ratio);
                    super.onMeasure(
                            widthMeasureSpec,
                            MeasureSpec.makeMeasureSpec(height, MeasureSpec.EXACTLY)
                    );
                    return;
                }
            **}else{
                super.onMeasure(widthMeasureSpec, heightMeasureSpec);
                return;
            }**
            
        }

        super.onMeasure(widthMeasureSpec, heightMeasureSpec);
    }

calebdre, 2017-03-09:
@android-dataticket you should make a pull request with that code

cears-bhumika, 2017-07-24:
@android-dataticket still i am getting same issue in some of the samsung devices.

closed 2018-02-03 by austinkettner