CameraKit/camerakit-android #19: Taking photos in landscape mode distorts preview

pmb516, 2017-03-09:


paxxux, 2017-03-11:
Hi,

I expierence the same! My activity is in landscape and the camera preview shows a weird angle. But the pictures are taken correctly.

Are you going to fix this soon? I really want to use this library, but I need a solution for this in the next days :(.

Please, please, please!

dwillmc, 2017-03-11:
I'll try to put a fix in for this tomorrow

paxxux, 2017-03-11:
awesome! many thanks!

dwillmc, 2017-03-12:
Looking into this now

pmb516, 2017-03-15:
Just updated to 0.9.15 and the preview is no longer distorted! Great work.

However, it's now 90 degrees out of rotation in landscape mode 😞

kevcgrant, 2017-03-20:
In landscape mode the preview is rotated 90 degrees for me as well.

pmb516, 2017-03-22:
@dwillmc Any update on this?

andreidiaconu, 2017-03-23:
I don't have a lot of devices but #39 is what worked for my device. @pmb516 @kevcgrant , since you also reported rotation issues, would you mind checking out this branch and trying it on your device? Thanks!

kevcgrant, 2017-03-23:
I will try tonight and post my findings

kevcgrant, 2017-03-24:
@andreidiaconu I checked out your branch and tested, everything works except if you use Still mode, the **output** is rotated 90 degrees. The preview is correct for both Standard and Still modes, and the output is correct for Standard as well

wangchengyan, 2017-04-06:
i need use SurfaceView the Preview still distorted  can you help me? thanks

wangchengyan, 2017-04-07:
when i set activity landscape 

in xml set CameraView
width="match_parent"
height="476px"

Preview still  distorted

dwillmc, 2017-04-17:
Now fixed, please update to `0.9.18`.

wangchengyan, 2017-04-18:
thanks very much!!

kuoliangkwong, 2017-09-18:
Hi im having the same distortion when I use android:configChanges="orientation|screenSize|keyboardHidden" in my Activity using 0.9.18. 

[image]
[image]

YeWin, 2017-12-13:
This issue is still not OK on 0.12.3 latest. Do you have any clue?

austinkettner, 2017-12-13:
I need to confirm, as this seems like a dup of another issue, but reOpened for now!

commit c1e42885 (2017-12-13): Use display orientation instead of device orientation for layouting preview

Resolves #19 Resolves #257
commit 5875fded (2017-12-13): Fix preview rotation issues

Resolves #257, Resolves #282, Resolves #19
PR #280 (2017-12-13): Release v0.13.1
Resolves #256 
Resolves #19 
Resolves #257 
Resolves #282 
Resolves #273
closed 2017-12-13 by dwillmc