CameraKit/camerakit-android #20: Crash on launching Front camera preview - Nexus 6

nethergrim, 2017-03-09:
Hi!

Facing next crash on Nexus 6, during start of camera preview.

Android 6.0.1 (stock firmware).

`Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=450
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3140)
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:347)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:25)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:60)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
       at android.view.TextureView.getHardwareLayer(TextureView.java:368)
       at android.view.View.updateDisplayListIfDirty(View.java:15157)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.updateDisplayListIfDirty(View.java:15175)
       at android.view.View.draw(View.java:15954)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
       at android.view.View.draw(View.java:16187)
       at com.android.internal.policy.PhoneWindow$DecorView.draw(PhoneWindow.java:2690)
       at android.view.View.updateDisplayListIfDirty(View.java:15180)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:281)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:287)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:322)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:2615)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2434)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2067)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1107)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6013)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:858)
       at android.view.Choreographer.doCallbacks(Choreographer.java:670)
       at android.view.Choreographer.doFrame(Choreographer.java:606)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.app.ActivityThread.main(ActivityThread.java:5417)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)`

Config:

        setFacing(CameraKit.Constants.FACING_FRONT);
        setFlash(CameraKit.Constants.FLASH_OFF);
        setFocus(CameraKit.Constants.FOCUS_CONTINUOUS);
        setMethod(CameraKit.Constants.METHOD_STILL);
        setCropOutput(true);
        setJpegQuality(100);

Version:

compile 'com.flurgle:camerakit:0.9.13'

nethergrim, 2017-03-09:
Please note that permission check is being performed before that, and application is granted with Camera permission.

suomi35, 2017-03-09:
I am seeing exactly the same behavior with v0.9.13 on a Genymotion VM (Nexus S, 4.1.1).
I am hunting for a way to properly try/catch the IllegalArgumentException, but have not yet found a good way, as the init seems to happen above the activity.

nethergrim, 2017-03-09:
Created a Pull Request to fix this issue: [link](https://github.com/flurgle/CameraKit-Android/pull/21)

closed 2017-03-09 by dwillmc