CameraKit/camerakit-android #25: Video Capture Issue

arundilipan, 2017-03-10:
While attempting to capture video through `onVideoCaptured` the file parameter `video` points to a directory that does not exist, and cannot be played through `VideoView`. When I try and access the video through `VideoView` it outputs a `java.io.FileNotFoundException`, and I'm not sure what is going on here.

dwillmc, 2017-03-10:
I'll look into this soon. Are you using Android M or above? Can you check if the storage permission is approved for your app.

arundilipan, 2017-03-11:
We're running this on Nougat, and we already enabled external storage permissions.

dwillmc, 2017-03-11:
Thanks for checking. Video is one of the shakier parts of the library right now. Are you attempting to show the video in VideoView right when you finish recording? If so can you try setting a delay of about 3 seconds before you attempt to use the file to see if that's the issue?

arundilipan, 2017-03-11:
We just tried doing that, and it delayed 3 seconds before throwing that exact same `FileNotFoundException`.

arundilipan, 2017-03-12:
I also found something slightly concerning, which may be the reason why we're having issues. I checked the `Camera2.java` file, and I found that many of the methods were unimplemented, including `endVideo()`. If I'm wrong please don't hesitate to correct me.

dwillmc, 2017-03-12:
Camera2 isn't used anywhere yet. It's just the base implementation which I'm commiting soon. All code in your app currently will be from Camera1.java (for the next few days at least until 1.0)

arundilipan, 2017-03-12:
Oh, my apologies then. Do you have any other suggestions as to what might be wrong with the `File` not being saved?

dwillmc, 2017-03-12:
Can you send me a bit of your implementation? `startVideo()` call, `endVideo()` call and the callback implementation

arundilipan, 2017-03-12:
```
package com.imad.bullitin.MainLayouts;

import android.app.Fragment;
import android.content.Intent;
import android.graphics.Bitmap;
import android.graphics.BitmapFactory;
import android.os.Bundle;
import android.support.design.widget.FloatingActionButton;
import android.view.LayoutInflater;
import android.view.View;
import android.view.ViewGroup;
import android.widget.ImageButton;
import android.widget.ImageView;
import android.widget.VideoView;

import com.flurgle.camerakit.CameraKit;
import com.flurgle.camerakit.CameraListener;
import com.flurgle.camerakit.CameraView;
import com.imad.bullitin.MediaUploadUtils.MediaUpload;
import com.imad.bullitin.R;

import java.io.File;

/**
 * Created by imadkhokher on 3/10/17.
 */

public class Camera extends android.support.v4.app.Fragment {

    CameraView camera;
    private ImageButton resetMedia;
    private ImageView imageView;
    private VideoView videoView;
    private boolean isReordingVideo;
    public static File videoFile;
    public static Bitmap bitmap;
    public static boolean capturedImage;
    public static FloatingActionButton imageCapture, videoCapture, postMedia;

    @Override
    public View onCreateView(final LayoutInflater inflater, ViewGroup container, Bundle savedInstanceState) {

        final View v = inflater.inflate(R.layout.camera_fragment, container, false);

        camera = (CameraView) v.findViewById(R.id.camera);
        resetMedia = (ImageButton) v.findViewById(R.id.reset_media);

        imageCapture = (FloatingActionButton) v.findViewById(R.id.camera_image);
        videoCapture = (FloatingActionButton) v.findViewById(R.id.camera_video);

        postMedia = (FloatingActionButton) v.findViewById(R.id.post_media);

        imageView = (ImageView) v.findViewById(R.id.imageView);
        videoView = (VideoView) v.findViewById(R.id.videoView);
        isReordingVideo = false;

        camera.setCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(byte[] picture) {
                super.onPictureTaken(picture);

                bitmap = BitmapFactory.decodeByteArray(picture, 0, picture.length);
                imageView.setImageBitmap(bitmap);

            }

            @Override
            public void onCameraOpened() {
                super.onCameraOpened();
[... 114 lines cut ...]
        camera.stop();
        super.onPause();
    }
    @Override
    public void onStart() {
        super.onStart();
    }

    @Override
    public void onStop() {
        super.onStop();
    }

    @Override
    public void onDestroy(){
        super.onDestroy();
    }

    @Override
    public void onActivityCreated(Bundle savedInstanceState) {
        super.onActivityCreated(savedInstanceState);

    }
}
```

dwillmc, 2017-03-12:
Can you send me your layout file too so I can easily run myself

arundilipan, 2017-03-12:
I don't think I can because you wouldn't be able to run it with the layout file we have. It's a layout for a fragment and this fragment is dependent on the rest of our project. You wouldn't be able to run it without the entire project.

dwillmc, 2017-03-12:
I've reproduced your problem, give me a few mins

dwillmc, 2017-03-12:
Instead of just:

```
videoView.setVideoPath(video.getAbsolutePath());
```

Do:

```
videoView.setVideoPath(video.getAbsolutePath());
videoView.setMediaController(null);
videoView.setOnPreparedListener(new MediaPlayer.OnPreparedListener() {
    @Override
    public void onPrepared(MediaPlayer mediaPlayer) {
        mediaPlayer.setLooping(true);
    }
});
videoView.start();
```

Let me know how it goes

dwillmc, 2017-03-12:
And no need for the delay before doing it that i suggested before

arundilipan, 2017-03-12:
So, we found that it works, but on Samsung devices, the bitmap produced by the camera is incorrectly oriented, which means the images are cropped with black bars.

dwillmc, 2017-04-03:
The original issue here is solved and your last comment has another issue so I'll close this!

closed 2017-04-03 by dwillmc