CameraKit/camerakit-android #34: RuntimeException in Tap to focus

xp-vit, 2017-03-17:
Thanks a lot for a great library! I mentioned that you have added Tap To focus and decided to test it. I was able to get RuntimeException thrown on my Nexus 4 running Android 5.0.1. When I tap rather fast on item which is really close to camera and then to item which is far away. I got following:

03-17 16:15:50.295 20219-20219/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.flurgle.camerakit.demo, PID: 20219
                                                   java.lang.RuntimeException: setParameters failed
                                                       at android.hardware.Camera.native_setParameters(Native Method)
                                                       at android.hardware.Camera.setParameters(Camera.java:1876)
                                                       at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:459)
                                                       at android.view.View.dispatchTouchEvent(View.java:8465)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2427)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2405)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2106)
                                                       at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2369)
                                                       at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1719)
                                                       at android.app.Activity.dispatchTouchEvent(Activity.java:2742)
                                                       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                       at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2330)
                                                       at android.view.View.dispatchPointerEvent(View.java:8666)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4123)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:3989)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3544)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3597)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3563)
                                                       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3680)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3571)
                                                       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3737)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3544)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3597)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3563)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3571)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3544)
                                                       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5807)
                                                       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5781)
                                                       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5752)
                                                       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:5897)
                                                       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                       at android.os.MessageQueue.nativePollOnce(Native Method)
                                                       at android.os.MessageQueue.next(MessageQueue.java:143)
                                                       at android.os.Looper.loop(Looper.java:122)
                                                       at android.app.ActivityThread.main(ActivityThread.java:5254)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at java.lang.reflect.Method.invoke(Method.java:372)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
                                                   	at com.android.internal.os.ZygoteInit.main(ZygoteInit.jav
                                                   
                                                   --------- beginning of system

2astm, 2017-05-05:
the same

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc