CameraKit/camerakit-android #36: Picture rotate on the Sumsung phone.

Forsaken-sl, 2017-03-23:


michael-oknov, 2017-03-24:
I have the same issue =(

canberkcakmak, 2017-04-07:
you have to get exif images then rotate image with this code. after save bitmap on taken picture. This is the best solution of rotated images.

```
        switch(orientation) {

            case ExifInterface.ORIENTATION_ROTATE_90:
                selectedBitmap =rotateImage(selectedBitmap, 90);
                break;

            case ExifInterface.ORIENTATION_ROTATE_180:
                selectedBitmap =  rotateImage(selectedBitmap, 180);
                break;

            case ExifInterface.ORIENTATION_ROTATE_270:
                selectedBitmap =  rotateImage(selectedBitmap, 270);
                break;

            case ExifInterface.ORIENTATION_NORMAL:
                selectedBitmap =  rotateImage(selectedBitmap, 0);

            default:
                break;
        }
```

Dewinster, 2017-04-13:
I had the same issue, but solved it by using a different method of taking a picture by calling this function in the code:

`cameraView.setMethod(CameraKit.Constants.METHOD_STILL);`

Or this in you layout file: 

`camerakit:ckMethod="still"`

smhk, 2017-05-09:
Dewinster, **that's an amazingly helpful tip.**

Thank you so much - I wish I could send money!  :)

`cameraView.setMethod(CameraKit.Constants.METHOD_STILL);`

closed 2017-05-11 by Forsaken-sl

hablema, 2017-05-25:
Hi,

`camerakit:ckMethod="still"`  is a nice way to solve the problem as the preview is returned but is there any other way so that i can use the flash which works only in standard mode when image is taken from camera ?

Kwonjube, 2019-04-03:
Has this issue been resolved? I'm still getting this problem with an samsung galaxy 10.1 (SM-T580) android 7.0
The output file is just randomly rotated 90 degree, i've fixed the screen orientation so the capture and preview screen are always correct
```
 if (jpeg != null) {
            imageView.setVisibility(View.VISIBLE);

            Bitmap bitmap = BitmapFactory.decodeByteArray(jpeg, 0, jpeg.length);
            if (bitmap == null) {
                finish();
                return;
            }

            imageView.setImageBitmap(bitmap);
            actualResolution.setText(bitmap.getWidth() + " x " + bitmap.getHeight());
            approxUncompressedSize.setText(getApproximateFileMegabytes(bitmap) + "MB");
            captureLatency.setText(ResultHolder.getTimeToCallback() + " milliseconds");
            File savedPhoto = new File(Environment.getExternalStorageDirectory(), "photo.jpg");
            try {
                FileOutputStream outputStream = new FileOutputStream(savedPhoto.getPath());
                outputStream.write(jpeg);
                outputStream.close();
            } catch (java.io.IOException e) {
                e.printStackTrace();
            }

        }
```