CameraKit/camerakit-android #38: Landscape preview and video capture

kevin-zqw, 2017-03-23:
Hi, I'm working on an app which support landscape only. I want to use this awesome library to capture video in landscape mode.

But it seems that this library doesn't support landscape activity, the preview was rotated and distorted.

How to fix this issue?

andreidiaconu, 2017-03-23:
Hi! I tried to fix a similar issue reported in the comments of #19 . Do you mind testing using #39 on your device? Thanks!

kevin-zqw, 2017-03-24:
@andreidiaconu I tested your pull request #39 , it works perfect on my two test devices.

andreidiaconu, 2017-03-24:
I am glad to hear that! There are still some issues with STILL mode, described in the PR comments on #39 which I will try to fix and hopefully the PR will be approved.

PR #39 (2017-04-17): Fixed landscape rotation issues
This fixes the orientation problems discussed on #19 (at least on my device - Nexus 6 with android 7.0) and might also fix #38

I removed the matrix transformation on the preview, as it was doing the same thing as `mCamera.setDisplayOrientation(calculateCameraRotation(mDisplayOrientation));` from `Camera1.java` - resulting in double-rotation in landscape.

If you wish to test, you need to remove `android:screenOrientation="portrait"` from the Manifest

The removed matrix transformation from `TextureViewPreview` seems like a lot of work and I imagine it covered a [... cut]
closed 2017-04-17 by dwillmc