CameraKit/camerakit-android #40: Only first 6 seconds of video are saved to storage

mbernr, 2017-03-23:
I tried on both the Nexus 5X (7.1) and the LG K8 (6.0). 

I do exactly as in the example:

`cameraView.startRecordingVideo();
        cameraView.postDelayed(new Runnable() {
            @Override
            public void run() {
                cameraView.stopRecordingVideo();
            }
        }, VIDEO_TIME_LIMIT);`

`onVideoTaken` gets called after VIDEO_TIME_LIMIT, just like I want it. But regardless, only the first 6 seconds of the video are saved to storage.

johntzan, 2017-03-23:
having the same issue here, only the first few seconds are recorded for playback.

johntzan, 2017-03-31:
any updates or workarounds for this?

this is what the logs look like while recording: 

03-30 21:47:14.121 28884-28884 I/ViewRootImpl: ViewRoot's Touch Event : ACTION_DOWN
03-30 21:47:14.125 28884-28884 V/MediaRecorder: constructor
03-30 21:47:14.133 28884-28884 D/MediaRecorder: mAudioZoomEnable = 1
03-30 21:47:14.133 28884-28884 V/MediaRecorder: doCleanUp
03-30 21:47:14.135 28884-28884 V/MediaRecorder: setListener
03-30 21:47:14.135 28884-28884 V/MediaRecorder: setClientName
03-30 21:47:14.139 28884-28884 V/MediaRecorder: setCamera(0x78b16fa800,0x78b16fae00)
03-30 21:47:14.142 28884-28884 V/MediaRecorder: setVideoSource(1)
03-30 21:47:14.142 28884-28884 V/MediaRecorder: Call init() since the media recorder is not initialized yet
03-30 21:47:14.142 28884-28884 V/MediaRecorder: init
03-30 21:47:14.149 28884-28884 V/MediaRecorder: setAudioSource(5)
03-30 21:47:14.155 28884-28884 V/MediaRecorder: setOutputFormat(2)
03-30 21:47:14.155 28884-28884 V/MediaRecorder: setVideoFrameRate(30)
03-30 21:47:14.156 28884-28884 V/MediaRecorder: setVideoSize(720, 480)
03-30 21:47:14.157 28884-28884 V/MediaRecorder: setVideoEncoder(2)
03-30 21:47:14.161 28884-28884 V/MediaRecorder: setAudioEncoder(3)
03-30 21:47:14.178 28884-28884 D/MediaRecorder: _setOutputFile E
03-30 21:47:14.178 28884-28884 V/MediaRecorder: setOutputFile(152, 0, 0)
03-30 21:47:14.180 28884-28884 D/MediaRecorder: _setOutputFile X
03-30 21:47:14.180 28884-28884 V/MediaRecorder: prepare
03-30 21:47:14.182 28884-28884 V/MediaRecorder: start
**03-30 21:47:14.919** 28884-28884 I/Choreographer: Skipped 47 frames!  The application may be doing too much work on its main thread.
03-30 21:47:21.753 28884-29024 V/MediaRecorder: message received msg=2, ext1=801, ext2=0
**03-30 21:47:21.753** 28884-29024 V/MediaRecorder: callback application
03-30 21:47:21.754 28884-29024 V/MediaRecorder: back from callback
03-30 21:47:21.754 28884-29024 V/MediaRecorder: message received msg=101, ext1=268436456, ext2=0
03-30 21:47:21.754 28884-29024 V/MediaRecorder: callback application
03-30 21:47:21.754 28884-29024 V/MediaRecorder: back from callback
03-30 21:47:21.783 28884-28898 V/MediaRecorder: message received msg=101, ext1=536871912, ext2=0
03-30 21:47:21.783 28884-28898 V/MediaRecorder: callback application
03-30 21:47:21.784 28884-28898 V/MediaRecorder: back from callback
03-30 21:47:26.807 28884-28884 I/ViewRootImpl: ViewRoot's Touch Event : ACTION_UP
**03-30 21:47:26.808** 28884-28884 V/MediaRecorder: stop
03-30 21:47:27.309 28884-28884 V/MediaRecorder: doCleanUp

at around 6 seconds highlighted above the callback is called. even though the recording keeps going for 12 seconds.

mishasrb, 2017-03-31:
Hi, problem is in https://github.com/gogopop/CameraKit-Android/blob/master/camerakit/src/main/api16/com/flurgle/camerakit/Camera1.java. Max file size is set to be 5MB and max duration is set to 20sec.
`mMediaRecorder.setMaxDuration(20000);`
`mMediaRecorder.setMaxFileSize(5000000);`

dwillmc, 2017-04-03:
This is fixed in `0.9.16` coming later today. Thanks for the issue!

closed 2017-04-03 by dwillmc