CameraKit/camerakit-android #41: Toggle facing breaks continuous focus

lukaspili, 2017-03-24:
Steps to reproduce: 
1. Start cameraview with focus set on continuous and facing set on back;
2. Toggle facing using `cameraView.toggleFacing()`;
3. Camera does not focus anymore, neither in back or front facing.

Calling `cameraView.setFocus(CameraKit.Constants.FOCUS_CONTINUOUS)` after toggling does not help.

jabalsick, 2017-03-27:
Hi! i have to put CameraKit.Constants.FOCUS_CONTINUOUS as a default Value to make it work.

lukaspili, 2017-03-27:
What do you mean?

jlmcdonnell, 2017-11-30:
Has the cause of this been found yet? I am also able to reproduce this issue on `0.12.1`

jlmcdonnell, 2017-11-30:
I found a temporary workaround to get focus working again after switching camera until the underlying issue is resolved.

After changing to the front or back camera:
```
cameraView.stop()
cameraView.start()
cameraView.setFocus(CameraKit.Constants.FOCUS_CONTINUOUS)
```

emersoncloud, 2018-07-30:
This issue has been resolved in v1.0.0. Closing this issue. If there still are focus issues in 1.0.0 I will make new ticket to keep track!

closed 2018-07-30 by emersoncloud