CameraKit/camerakit-android #42: Pinch to zoom is not working

911shanky, 2017-03-25:
It seems like "Pinch to Zoom" is not working.

Calling ` camera.setZoom(CameraKit.Constants.ZOOM_PINCH);`  doesn't work.

Can you resolve this issue quickly.

Moonbloom, 2017-03-25:
This is not implemented yet, so there is nothing to "resolve".

If you read the README, you'll see:

Coming soon: Built-in pinch to zoom.

The code to enable the option is in the lib, but the functionality is not there yet.

android-dataticket, 2017-04-18:
having huge client demand for this feature, hope it gets implemented soon

android-dataticket, 2017-04-20:
i added pinch to zoom, pull #72

veeranath, 2017-06-05:
Hi sir i am using this version of the library com.flurgle:camerakit:0.9.17 is pinch zoom is intergated as tried it .Its not working sir

veeranath, 2017-06-05:
@android-dataticket  i had tried your pull request but pinch to zoom is not working

samuelhehe, 2017-07-17:
@android-dataticket i had tried your pull request but pinch to zoom is not working

adamwlarson, 2017-07-27:
I took @android-dataticket pull request and integrated into my fork, I had to do a few things to get it working such as passing on the onTouchEvent, I also switched over to a gesture scale. It isn't working perfectly yet, but it is close. I am using jitpack and have a release tagged https://github.com/adamwlarson/CameraKit-Android/releases/tag/1.0.4

android-dataticket, 2017-07-27:
sorry I've sorta abandoned this thread @adamwlarson - I made the pinch to zoom pull request and then had to make some more adjustments to get it to work and never updated the PR

looks like you figured it out, i think the creator of this library has abandoned it.. i gave up.

i have a working fork that i use in my app and haven't updated it in months, it gives me some memory issues here and there but it works for the most part, i wish this library was a little more up-kept by the creator but oh well, he's a busy young man i guess

KalpeshJadvani, 2017-08-12:
any one to solved this problem  plz tell me i was finding from long time

abhisheksingapore, 2017-10-27:
Has anyone solved this issue, yet? Pinch to zoom is critical to my app and everything else about this fits my app well except the lack of zoom

android-dataticket, 2017-10-27:
@abhisheksingapore @KalpeshJadvani I've updated the pull request with some changes that I made afterwards which work for my application

aravindhkumar23, 2017-11-03:
@android-dataticket please confirm when can i get the pinch to zoom feature?

austinkettner, 2017-11-03:
@android-dataticket Can you reOpen the PR? @aravindhkumar23 this is something we are working on, you can expect it in the next week.

android-dataticket, 2017-11-03:
@austinkettner i am not that great of a programmer and what i've done is forked and tweaked to work with only my application.. for instance consider the following:
in my PR, i remove DisplayOrientationDetector - some may need this, but i did not
in my PR, i removed the handleFocus(event, params) method call, it gave issues when touching the screen to pinch
i also add some quirky try catches and thread sleeps, but if you really want me to do a fresh PR, i will fork a new copy and upload the files that i've changed

austinkettner, 2017-11-03:
@android-dataticket No worries! Thanks for the clarity, we are actually working on a solution to this today internally, so no PR needed! Really appreciate your willingness to help though! =)

aravindhkumar23, 2017-11-09:
Hi @austinkettner any updates regarding pinch to zoom feature? or any workaround to get it done?
i have been waiting for this feature to be implemented in one of my project..

austinkettner, 2017-11-09:
@aravindhkumar23 We actually just pushed this in the `release/0.13.0` branch (https://github.com/wonderkiln/CameraKit-Android/tree/release/0.13.0), check it out via `compile 'com.wonderkiln:camerakit:0.13.0-Snapshot'	`

aravindhkumar23, 2017-11-11:
when i use `compile 'com.wonderkiln:camerakit:0.13.0-Snapshot'` 
i get error msg when building gradle
> Failed to resolve: com.wonderkiln:camerakit:0.13.0-Snapshot

when i use `compile 'com.wonderkiln:camerakit:0.12.0'` the gradle builds successfully
but i face issue in using listener events
> cannot find symbol class CameraListener

my app build.gradle is here

```
android {
    compileSdkVersion 27
    buildToolsVersion '27'
    defaultConfig {
        applicationId "com.smthing.nothing"
        minSdkVersion 16
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"
        vectorDrawables.useSupportLibrary = true
        dataBinding {
            enabled = true
        }
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
}

dependencies {
    compile fileTree(dir: 'libs', include: ['*.jar'])
    androidTestCompile('com.android.support.test.espresso:espresso-core:2.2.2', {
        exclude group: 'com.android.support', module: 'support-annotations'
    })
    compile 'com.android.support:appcompat-v7:27.0.0'
    compile 'com.android.support.constraint:constraint-layout:1.0.2'
    compile 'com.android.support:design:27.0.0'
    compile 'com.github.yalantis:ucrop:2.2.1'
    compile 'com.wonderkiln:camerakit:0.12.0'
}
```

--i'am a newbie in android app development can u guys help me to solve this

austinkettner, 2017-11-11:
There is a temporary issue with Bintray current preventing the builds from being accessible, you can use `compile 'com.wonderkiln:camerakit:0.13.0-Snapshot'` in the meantime. @aravindhkumar23

aravindhkumar23, 2017-11-13:
@austinkettner `compile 'com.flurgle:camerakit:0.13.0-Snapshot'` zoom works but the listener part shows error on building and running the project,

> cannot find symbol class CameraListener
> cannot resolve method setCameraListener()

emersoncloud, 2018-07-30:
Pinch to zoom and capture image have been resolved in 1.0.0. Closing this issue.

closed 2018-07-30 by emersoncloud