CameraKit/camerakit-android #48: java.lang.RuntimeException: Method called after release()

dayanithi06, 2017-03-30:
java.lang.RuntimeException: Method called after release()
                                                                    at android.hardware.Camera.native_setParameters(Native Method)
                                                                    at android.hardware.Camera.setParameters(Camera.java:2550)
                                                                    at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:136)
                                                                    at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:367)
                                                                    at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
                                                                    at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:68)
                                                                    at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                                    at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
                                                                    at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
                                                                    at android.view.TextureView.getHardwareLayer(TextureView.java:385)
                                                                    at android.view.View.getDisplayList(View.java:13554)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.draw(View.java:14745)
                                                                    at android.widget.FrameLayout.draw(FrameLayout.java:534)
                                                                    at android.view.View.getDisplayList(View.java:13594)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at android.view.View.getDisplayList(View.java:13586)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.View.draw(View.java:14436)
                                                                    at android.view.ViewGroup.drawChild(ViewGroup.java:3252)
                                                                    at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3070)
                                                                    at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchDraw(PhoneWindow.java:2387)
                                                                    at android.view.View.draw(View.java:14745)
                                                                    at android.widget.FrameLayout.draw(FrameLayout.java:534)
                                                                    at com.android.internal.policy.impl.PhoneWindow$DecorView.draw(PhoneWindow.java:2367)
                                                                    at android.view.View.getDisplayList(View.java:13594)
                                                                    at android.view.View.getDisplayList(View.java:13641)
                                                                    at android.view.HardwareRenderer$GlRenderer.buildDisplayList(HardwareRenderer.java:1627)
                                                                    at android.view.HardwareRenderer$GlRenderer.draw(HardwareRenderer.java:1506)
                                                                    at android.view.ViewRootImpl.draw(ViewRootImpl.java:2722)
                                                                    at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2587)
                                                                    at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2210)
                                                                    at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1189)
                                                                    at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6223)
                                                                	at android.view.Choreographer$Call

sw-tt-rahulshah, 2017-06-22:
Have you got any solution for this issue?

dayanithi06, 2017-06-22:
@Override
    protected void onDestroy() {
        super.onDestroy();
        Handler handler = new Handler();
        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                if (cameraView != null) {
                    //  cameraView.stopRecordingVideo();
                    cameraView.stop();
                }

            }
        }, 1000);

    }

Try this .. This one fixed the issue for me.

sw-tt-rahulshah, 2017-06-22:
Thanks for the reply but it doesnt work for me. Have you written any code in onPause() ?

sw-tt-rahulshah, 2017-06-22:
Actually I'm frequently opening and closing camera screen and this crash comes after 3rd or 4th time. May be you might still be getting this issue.

dayanithi06, 2017-06-22:
@Override
    protected void onResume() {
        super.onResume();
        if (cameraView != null) {
            cameraView.setAlpha(1);
            Handler handler = new Handler();
            handler.postDelayed(new Runnable() {
                @Override
                public void run() {
                    if (cameraView != null) {
                        //  cameraView.stopRecordingVideo();
                        cameraView.start();
                    }
                }
            }, 1000);  
      
}

       

   

    @Override
    protected void onPause() {
        Handler handler = new Handler();
        if (cameraView != null) {

            handler.postDelayed(new Runnable() {
                @Override
                public void run() {
                    if (cameraView != null) {
                        //  cameraView.stopRecordingVideo();
                        cameraView.stop();
                    }

                }
            }, 1000);
        }

    
        // cameraView.getHolder().removeCallback(cameraView);
      
        super.onPause();

    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        Handler handler = new Handler();
        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                if (cameraView != null) {
                    //  cameraView.stopRecordingVideo();
                    cameraView.stop();
                }

            }
        }, 1000);

    }

sw-tt-rahulshah, 2017-06-22:
It still doesn't help when you move screen from foreground to background and vice versa frequently. :-(

dayanithi06, 2017-06-22:
Post your  code.. Lets check can we improve it.

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc