CameraKit/camerakit-android #49: ConcurrentModificationException when starting camera

mcassiano, 2017-03-31:
Sometimes the library will throw this error when starting. My setting is:
Fragment 1 (No camera view), Fragment 2 (Camera View).

When you transit between these fragments it works just fine and stops/starts camera, but eventually it will crash the app. Might be related to not killing handlers properly when stopping?

Stacktrace:

```
03-31 09:48:01.920 22438-23468/[REDACTED] E/AndroidRuntime: FATAL EXCEPTION: Thread-1326
      Process: [REDACTED], PID: 22438
      java.util.ConcurrentModificationException
          at java.util.LinkedHashMap$LinkedHashIterator.nextEntry(LinkedHashMap.java:346)
          at java.util.LinkedHashMap$KeyIterator.next(LinkedHashMap.java:366)
          at android.hardware.Camera$Parameters.flatten(Camera.java:2830)
          at android.hardware.Camera.setParameters(Camera.java:2019)
          at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:136)
          at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:367)
          at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:308)
          at com.flurgle.camerakit.Camera1.start(Camera1.java:82)
          at com.flurgle.camerakit.CameraView$2.run(CameraView.java:160)
          at java.lang.Thread.run(Thread.java:818)
```

Dewinster, 2017-04-04:
This problem occurs on activities as well.
Maybe the problem is that the process is not running on the main-thread?

The problem did not appear to me anymore when I moved:
`mCameraImpl.start();`

out of 

`new Thread(new Runnable() {
            @Override
            public void run() {
                     mCameraImpl.start();
            }
        }).start();`

Not sure if this is a correct fix, or if it fixing the problem at all.
I am not really into Android's camera. 😄

mcassiano, 2017-04-04:
@Dewinster heh, that might actually be the issue. Does it happen with takePicture() too? I was running it in the I/O Scheduler.

Dewinster, 2017-04-04:
@mcassiano It doesn't happen with the takePicture() for me.

mcassiano, 2017-04-04:
@Dewinster hmm, interesting. well it seems to be sorted out for a while now, so imma close this issue. thanks man!

closed 2017-04-04 by mcassiano

snowpong, 2017-04-20:
This crash has been happening to me all day after I upgraded from 0.9.13 to 0.9.18

It happens as I open the Fragment containing the CameraView.
It happens around 30% of the time.

I'm testing on a Huawei Mate 9 Pro running 7.0

```javascript
E: FATAL EXCEPTION: Thread-18
   Process: no.dnt.sjekkut, PID: 9705
   java.util.ConcurrentModificationException
       at java.util.LinkedHashMap$LinkedHashIterator.nextEntry(LinkedHashMap.java:411)
       at java.util.LinkedHashMap$KeyIterator.next(LinkedHashMap.java:422)
       at android.hardware.Camera$Parameters.flatten(Camera.java:3259)
       at android.hardware.Camera.setParameters(Camera.java:2586)
       at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:144)
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:389)
       at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:322)
       at com.flurgle.camerakit.Camera1.start(Camera1.java:89)
       at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
       at java.lang.Thread.run(Thread.java:776)
```

Dewinster, 2017-04-20:
Hi Snowpong,

This issue is currently not fixed in the gradle version. To fixed this issues you should import this project as a library and make the changes by yourself.

markGilchristBookingBug, 2017-05-05:
I am getting this crash with 1 in 100 users, are you going to upgrade the gradle version any time soon? or should I do as suggested above and import the project and fix?

Dewinster, 2017-05-05:
You should implement the fix and create a PR 😜

markGilchristBookingBug, 2017-05-06:
ok I will ask work they may well let me do it on there time

markGilchristBookingBug, 2017-05-07:
@Dewinster  I have cloned the project, and remove the code that starts the camera on a new thread, I am testing this now, do you want a PR for this ? I was looking at at issues #57 and I think that @android-dataticket 's solution maybe a better implementation. Is @android-dataticket  going to do a pr?

android-dataticket, 2017-05-08:
@markGilchristBookingBug i'm flattered that you're interested in my hack-up solution, but i am not planning on doing a pr for this.  it's just a custom implementation that i came up with to solve my issue as i load this camera view in a fragment inside a viewpager.. my solution is actually added to the fragment that holds the cameraview reference rather than the library itself.  i found that between some older devices my clients use, i needed just a little bit of a delay to let things catch up before starting the camera.  i did add pinch to zoom with a pr, but i also had to make modifications to the viewpager to avoid onTouch interception conflicts/crashes.. i dont think i've followed the library since version .15, it's just been my own local fork that i keep modifying.. i dont know where dylan is either, he seems to be away or otherwise abandoned this project temporarily

TheLester, 2017-06-23:
Issue is still present in 0.9.17 . Why its closed ?

JakeWoki, 2017-07-11:
0.9.17

Device: Honor 8 Lite
OS: EMUI5.0(Android 7.0)