CameraKit/camerakit-android #50: can I set default values before camera starting programically?

dphans, 2017-04-01:
If I need to set default camera is front before camera start, can I do that?

mbernr, 2017-04-18:
Please take a look at the README:

`cameraView.setFacing(CameraKit.Constants.FACING_FRONT);`

luongvo, 2017-04-19:
@dphans Take a look at all public methods inside `CameraView` object.

dphans, 2017-04-25:
@mbernr @luongvo I see. But when call `startCamera` method after `setFacing`, camera facing not correct update.
I've fixed temporary by this way (set default in layout file):
- In layout file: add 2 CameraView elements, one CameraView setup it default front-facing, one for back-facing.
- In activity: get user settings and select one in 2 CameraView to show on screen.

I think this is temporary fixed for company project, but can close this issue. Many thanks.

closed 2018-02-03 by austinkettner