CameraKit/camerakit-android #55: Get camerapreview frames

grupoestudoscds, 2017-04-06:
First thanks for this awesome library.

Second, I'm trying to use this library with google vision barcode api, and I'm stucking on how to get the camera preview's frames for the barcode processor. Taking pictures in a Runnable and processing on the onPictureTaken event slows down my app, and sometimes it crashes the Runnable (even using try...catch).

There is a way of hooking into the camera view and get the frames? Or if I can't do that, could you give some adivice on how can I achive this, and maybe making a fork and implementing it?

beiliubei, 2017-06-20:
@grupoestudoscds you can try https://github.com/beiliubei/CameraKit-Android  

which  setPreviewCallback

morejump, 2017-10-16:
can not see it in project @beiliubei

RubenSandwich, 2018-01-08:
@beiliubei Thank you for doing that work? Will you be trying to merge this work into the main repo?

closed 2018-02-03 by austinkettner

LivioGama, 2018-03-01:
Can't use this lib just for that. Too bad.

david-alejandro-reyes-milian, 2018-09-19:
Good job with the lib, yet very bad that there is no access to the frames in order to enable the Barcode scanner. Is there any way of improving it? I could implement it if someone provides a general idea.

emersoncloud, 2018-09-20:
Hey @david-alejandro-reyes-milian we are aware that getting access to the frames is a pretty big issue for a lot of developers. Issue #342 covers individual frame inspection, if you have any more thoughts or suggestions drop them there.

Currently our development attention has been focused on the official v1.0.0 release with stable photo and video support. However look for frame inspection in a future release!

XinyueZ, 2018-09-20:
there is already API ```override fun onFrame(cv: CameraKitView?, bytes: ByteArray?) ``` , but it doesn't work.
see ```FrameCallback``` in ```https://dl.bintray.com/camerakit/camerakit-android-snapshots/com/camerakit/camerakit/1.0.0-Snapshot/```
The set of this callback is empty.

wenq1, 2019-03-13:
Why is the issue closed before it is resolved?

xperiafan13-rom, 2019-04-23:
some here, I really need this

Zaniyar, 2019-05-25:
is this still not solved?