CameraKit/camerakit-android #61: Crash when tap to focus

hi2hcs, 2017-04-11:
`E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.flurgle.camerakit.demo, PID: 16874
                                                   java.lang.RuntimeException: setParameters failed
                                                       at android.hardware.Camera.native_setParameters(Native Method)
                                                       at android.hardware.Camera.setParameters(Camera.java:3028)
                                                       at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:515)
                                                       at android.view.View.dispatchTouchEvent(View.java:9422)
                                                       at com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:131)
                                                       at android.view.View.dispatchTouchEvent(View.java:9422)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2660)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2304)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2666)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2318)
                                                       at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2490)
                                                       at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1783)
                                                       at android.app.Activity.dispatchTouchEvent(Activity.java:2818)
                                                       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                       at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2451)
                                                       at android.view.View.dispatchPointerEvent(View.java:9668)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4870)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4728)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4233)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4286)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4252)
                                                       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4398)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4260)
                                                       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4455)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4233)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4286)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4252)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4260)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4233)
                                                       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6698)
                                                       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6672)
                                                       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6604)
                                                       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6893)
                                                       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                       at android.os.MessageQueue.nativePollOnce(Native Method)
                                                       at android.os.MessageQueue.next(MessageQueue.java:328)
                                                       at android.os.Looper.loop(Looper.java:164)`

DmitryMtv, 2017-04-17:
I have a similar error. Occurs only at the top of the screen tap.
I think the reason is that at a large screen resolution (>2000), the center is incorrectly calculated in the method calculateFocusArea .

luongvo, 2017-04-19:
I have the same problem too.

xiaqiuming, 2017-04-21:
private Rect calculateFocusArea(float x, float y) {
        int centerX = clamp(Float.valueOf((x / mPreview.getView().getWidth()) * 2000 - 1000).intValue(), getFocusAreaSize());
        int centerY = clamp(Float.valueOf((y / mPreview.getView().getHeight()) * 2000 - 1000).intValue(), getFocusAreaSize());
        return new Rect(
                centerX - getFocusAreaSize() / 2,
                centerY - getFocusAreaSize() / 2,
                centerX + getFocusAreaSize() / 2,
                centerY + getFocusAreaSize() / 2
        );
    }
change to ------>
    private Rect calculateFocusArea(float x, float y) {

        int newY = (int) (2000 - x /  mPreview.getView().getWidth() * 2000) - 1000;
        int newX = (int) (y /  mPreview.getView().getHeight() * 2000) - 1000;

        int left = newX - 60;
        int top = newY - 60;
        int right = newX + 60;
        int bottom = newY + 60;
        left = left < -1000 ? -1000 : left;
        top = top < -1000 ? -1000 : top;
        right = right > 1000 ? 1000 : right;
        bottom = bottom > 1000 ? 1000 : bottom;
        return new Rect(left, top, right, bottom);

    }

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc

dayburak, 2018-05-18:
I have the version 0.13 + still having the crash