CameraKit/camerakit-android #64: Is it possible to specify Output File Object manually?

tigranbs, 2017-04-13:
Thanks for super easy library, it's really awesome!
But after having some app built around this library we have a requirement to specify Output File Object which would be the output for recorded video. Is there a way to do this ?
I couldn't find it from existing API

dwillmc, 2017-04-13:
Next release, available in a day or 2.

ayubbaba, 2017-06-06:
Is it available in the latest Api ? i had searched it haven't found it in this API

tigranbs, 2017-06-06:
@ayubbaba I've added that feature when I saw it is missing 😕
So you can get it by pulling here
https://github.com/gogopop/CameraKit-Android/pull/70
Don't know exactly why, but pull request not merged ... that code working in our mobile apps now

ayubbaba, 2017-06-06:
I will pull the code but how can i intergate it can give me example like cameraview.setVideoOutputFile(Here desired folder right) ?

tigranbs, 2017-06-06:
API is the same, I just added another function overload for `File` argument
`void setVideoOutputFile(File outputFile)`
So the API would be something like this
`cameraview.setVideoOutputFile(fd)` where **fd** is your manually created file handler with `File` type (or extended from it).

I've used this functionality for making `Pipe Socket`, but for files it will work just fine.

ayubbaba, 2017-06-06:
Ok Thanks i will try

KingOfSu, 2017-07-05:
@ayubbaba   Excuse me , i use author's API  but my outputfile  don't have the mp4 file, and i want to compress the file by h264 but when i use it the project is error,  can you help me ? thanks

vishalpatel1327, 2017-08-22:
@tigranbs 

How can i get this method I just found 0.9.18 version but its havnt anything inside 

so can u plz tell me what to do for that?

tigranbs, 2017-08-22:
@vishalpatel1327 please take a look on a comment
https://github.com/gogopop/CameraKit-Android/issues/64#issuecomment-306577250
Don't know that match about versions, I just needed at that time this functionality, added it and asked for a pull request, but it's didn't merged to master. So you can just get files/changes from that pull request and edit the version that you have.

vishalpatel1327, 2017-08-22:
that means i cant use compiled lib 

I have to pull whole code and change those  file 

am i right @tigranbs

tigranbs, 2017-08-22:
@vishalpatel1327 probably Yes that's the only solution for this issue, I've did the same thing ... otherwise you can just talk with this repository maintainers and understand why they don't want to add that functionality

vishalpatel1327, 2017-08-22:
ohk Thank you so much for Giving such quick reply  :)

closed 2018-02-03 by austinkettner