CameraKit/camerakit-android #71: Tablets with natural landscape orientation causes distorted/rotated preview and images

Moonbloom, 2017-04-18:
I've tested with many phones and tablets, on both the 0.9.17 and 0.9.18 versions (there were some changes to orientation calculations).

The issue happens on both a Google Nexus 10 tablet, and a Samsung Galaxy Tab 4. No other tablets or phones have this issue.

When using both front and rear camera, the preview is distorted, and when an image is taken, it's distorted and incorrectly rotated  (even after rotating it according to the EXIF data).

The only similarity/difference i can see is the fact that these two tablets are naturally landscape oriented.
All phones and tablets that are naturally portrait oriented behave correctly.

victor5171, 2017-04-20:
This issue is happening to me on Moto X 2nd generation. If the activity containing de cameraView is in portrait view everything is OK. But when i rotate the device to landscape or start the Activity in landscape the camera preview is completely messed up.

EnduroXPro, 2017-04-20:
I am facing that problem on Samsung Galaxy S7

odinnou, 2017-04-25:
happen on One Plus 3 too :(

mjf009, 2017-04-30:
I had a similar issue and have found that creating a Landscape Layout of activity_main has fixed the issue for me. Preview of the camera in portrait and landscape looks fine now.

kemsky, 2017-05-02:
It seems that actual sensor orientation is not taken into account:

com.flurgle.camerakit.Camera1#adjustCameraParameters
```java
        ...
        Camera.CameraInfo cameraInfo = new Camera.CameraInfo();
        Camera.getCameraInfo(Camera.CameraInfo.CAMERA_FACING_BACK, cameraInfo);

        Log.e("camera", "orientation: " + cameraInfo.orientation);

        invertPreviewSizes = cameraInfo.orientation == 0 ? !invertPreviewSizes : invertPreviewSizes;
        ...
```

L3K0V, 2017-05-19:
So guys, did somebody found resolution of this issue?

matheuslazarin, 2017-06-09:
I am having the same problem. If someone has the solution, let us know

closed 2018-02-03 by austinkettner