CameraKit/camerakit-android #73: Landscape preview distorted / zoomed

mjf009, 2017-04-21:
Tested on Samsung tab and phone and I have a problem with when rotating to landscape the camera image looks zoomed in. It is fine on portrait mode and the captured preview is displaying ok on both landscape and portrait. Is there a fix for this.

kemsky, 2017-05-02:
Galaxy Tab 2 10.1, the same story.

rogerdcarvalho, 2017-10-30:
Anyone figured out how to fix this?

getupcast, 2017-11-04:
I've been able to resolve this. In Camera1.Java, line 14:

`this.mDeviceOrientation = deviceOrientation`

Is not always set correctly. The device arbitrarily sets it to 0, even if the device is rotated. I decided to create a new boolean to store whether or not the device is in landscape or not, and base this on displayOrientation, which seems to be set consistently:

```
   //Device orientation is not reliable. Create a separate boolean to keep track of real orientation
        if (mDisplayOrientation == 90 || mDisplayOrientation == 270) {
            isLandscape = true;
        }

```
Later in line 473, the previewSize is inverted based on the device orientation:

```
 boolean invertPreviewSizes = (mCameraInfo.orientation + mDeviceOrientation) % 180 == 90;
        if (mPreviewSize != null && invertPreviewSizes) {
            return new Size(mPreviewSize.getHeight(), mPreviewSize.getWidth());
        }
```

I switched this to my new boolean: 

```
boolean invertPreviewSizes = !isLandscape;
        if (mPreviewSize != null && invertPreviewSizes) {
            return new Size(mPreviewSize.getHeight(), mPreviewSize.getWidth());
        }
```
Secondly, on line 616, the mPreview previewParameters are also set based on the deviceOrientation, but on top of that, invertPreviewSizes is used to flip height and width: 

```
 if (mDeviceOrientation == 0 || mDeviceOrientation == 180) {
                mPreview.setPreviewParameters(
                        getPreviewResolution().getWidth(),
                        getPreviewResolution().getHeight(),
                        mCameraParameters.getPreviewFormat()
                );
            } else {
                mPreview.setPreviewParameters(
                        getPreviewResolution().getHeight(),
                        getPreviewResolution().getWidth(),
                        mCameraParameters.getPreviewFormat()
                );
            }

            mCameraParameters.setPreviewSize(
                    invertPreviewSizes ? getPreviewResolution().getHeight() : getPreviewResolution().getWidth(),
                    invertPreviewSizes ? getPreviewResolution().getWidth() : getPreviewResolution().getHeight()
            );
```
Here I also switched to my new boolean instead of mDeviceOrientation to check what is required, but I only use this information to determine whether to invert height and weight. I don't think it makes sense to first set the preview on width/height or vice/versa, but then invert again. So the new code ended up being:

```
boolean invertPreviewSizes = !isLandscape;

if (getPreviewResolution() != null) {

            mPreview.setPreviewParameters(
                    getPreviewResolution().getWidth(),
                    getPreviewResolution().getHeight(),
                    mCameraParameters.getPreviewFormat()
            );

            mCameraParameters.setPreviewSize(
                    invertPreviewSizes ? getPreviewResolution().getHeight() : getPreviewResolution().getWidth(),
                    invertPreviewSizes ? getPreviewResolution().getWidth() : getPreviewResolution().getHeight()
            );

```

Tested on Samsung S8 Android 7.0 and preview shows correctly in both Landscape and Portrait. Hope this helps to resolve this across the board for all devices.

austinkettner, 2017-11-04:
@getupcast Would you be willing to submit a Pull Request for this?

rogerdcarvalho, 2017-11-04:
I've been able to resolve this. In Camera1.Java, line 14:

`this.mDeviceOrientation = deviceOrientation`

Is not always set correctly. The device arbitrarily sets it to 0, even if the device is rotated. I decided to create a new boolean to store whether or not the device is in landscape or not, and base this on displayOrientation, which seems to be set consistently:

```
   //Device orientation is not reliable. Create a separate boolean to keep track of real orientation
        if (mDisplayOrientation == 90 || mDisplayOrientation == 270) {
            isLandscape = true;
        }

```
Later in line 473, the previewSize is inverted based on the device orientation:

```
 boolean invertPreviewSizes = (mCameraInfo.orientation + mDeviceOrientation) % 180 == 90;
        if (mPreviewSize != null && invertPreviewSizes) {
            return new Size(mPreviewSize.getHeight(), mPreviewSize.getWidth());
        }
```

I switched this to my new boolean: 

```
boolean invertPreviewSizes = !isLandscape;
        if (mPreviewSize != null && invertPreviewSizes) {
            return new Size(mPreviewSize.getHeight(), mPreviewSize.getWidth());
        }
```
Secondly, on line 616, the mPreview previewParameters are also set based on the deviceOrientation, but on top of that, invertPreviewSizes is used to flip height and width: 

```
 if (mDeviceOrientation == 0 || mDeviceOrientation == 180) {
                mPreview.setPreviewParameters(
                        getPreviewResolution().getWidth(),
                        getPreviewResolution().getHeight(),
                        mCameraParameters.getPreviewFormat()
                );
            } else {
                mPreview.setPreviewParameters(
                        getPreviewResolution().getHeight(),
                        getPreviewResolution().getWidth(),
                        mCameraParameters.getPreviewFormat()
                );
            }

            mCameraParameters.setPreviewSize(
                    invertPreviewSizes ? getPreviewResolution().getHeight() : getPreviewResolution().getWidth(),
                    invertPreviewSizes ? getPreviewResolution().getWidth() : getPreviewResolution().getHeight()
            );
```
Here I also switched to my new boolean instead of mDeviceOrientation to check what is required, but I only use this information to determine whether to invert height and weight. I don't think it makes sense to first set the preview on width/height or vice/versa, but then invert again. So the new code ended up being:

```
boolean invertPreviewSizes = !isLandscape;

if (getPreviewResolution() != null) {

            mPreview.setPreviewParameters(
                    getPreviewResolution().getWidth(),
                    getPreviewResolution().getHeight(),
                    mCameraParameters.getPreviewFormat()
            );

            mCameraParameters.setPreviewSize(
                    invertPreviewSizes ? getPreviewResolution().getHeight() : getPreviewResolution().getWidth(),
                    invertPreviewSizes ? getPreviewResolution().getWidth() : getPreviewResolution().getHeight()
            );

```

Tested on Samsung S8 Android 7.0 and preview shows correctly in both Landscape and Portrait. Hope this helps to resolve this across the board for all devices.

rogerdcarvalho, 2017-11-04:
@austinkettner sorry was logged in under corporate account. I can't really do a pull request because I made many other changes specific to my implementation that may muddy the waters....

austinkettner, 2017-11-04:
Understood @rogerdcarvalho we will get this in soon =)

harishvishwakarma, 2018-02-22:
Is this in the code yet? I am getting the issue on Samsung

austinkettner, 2018-02-22:
@harishvishwakarma Can you test this in the 1.0.0 PR? https://github.com/CameraKit/camerakit-android/pull/318 - should be resolved, but if not let me know!

harishvishwakarma, 2018-02-23:
@austinkettner : It is resolved in the mentioned PR. I am very much impressed at how much easy this makes Camera implementation. Thanks for the library. 
When are you planning to put this to master branch? I am planning to use it for a production application.
 Methods have changed in this one though and cannot really find documentation for that. 
1. How to turn on autofocus?
2. How to turn on auto-crop to view bounds?
3. How to turn on pinch to zoom?

austinkettner, 2018-02-23:
Glad it's working much better in the 1.0.0! Can you please comment in the PR with 1-3, we will add documentation for this shortly.

Documentation as a whole for 1.0.0 is still coming, so we plan on having a fully dedicated documentation site ahead of merging.

closed 2018-02-26 by austinkettner