CameraKit/camerakit-android #75: video recorded in landscape mode not being read as landscape

palaciosyansro, 2017-04-24:
Hi! I'm trying to make my camera able to record videos in portrait and landscape mode. I haven't been able to achieve that yet. When recording in landscape, the video twists in this unusual way, and I haven't found any parameter to set it as landscape.
Wonder if any of you dealed with this issue. 
Just for the record, I've read EVERY issue here and none of those contain any solution for my prob.
I'm also attaching some screenshots to show you how it behaves.
[image]

Thanks!

L3K0V, 2017-05-20:
I have same issue. Is there any workaround or something like this?

eomine, 2017-06-09:
I had the same problem. See my PR: #111

closed 2018-02-03 by austinkettner