CameraKit/camerakit-android #83: getParameters failed (empty parameters)

2astm, 2017-05-04:
at android.hardware.Camera.native_getParameters(Native Method)
                                                                            at android.hardware.Camera.getParameters(Camera.java:2011)
                                                                            at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:315)
                                                                            at com.flurgle.camerakit.Camera1.start(Camera1.java:85)
                                                                            at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)

closed 2017-05-04 by 2astm

Forsaken-sl, 2017-06-29:
hi, do you solve this already? can you please tell me how to fix this?

cotfas, 2017-07-30:
Any news about this issue?
I`m getting sometimes this error:

java.lang.RuntimeException: getParameters failed (empty parameters)

xp-vit, 2017-07-30:
I believe that this is happening because of starting camera in separate thread:
`sWorkerHandler.post(new Runnable() {
            @Override
            public void run() {
                mCameraImpl.start();
            }
        });`

I have replaced it in my code with:
` try {                    
     mCameraImpl.start(); 
 } catch (Exception ex) { 
     mCameraImpl.stop();  
 }`

And no more issues are happening.

cotfas, 2017-07-31:
So, there will be an update to the library?

I`m starting the camera as:

From  public void onResume() (or/and onViewCreated fragment)

as:

       new Handler().postDelayed(new Runnable() {
            @Override
            public void run() {
                 try {
            if (cameraView != null) {
                cameraView.setVisibility(View.VISIBLE);
                cameraView.start();
            }
        } catch (Exception ex) {
        }
            }
        }, 500);

and onPause

   try {
            if (cameraView != null) {
                cameraView.setVisibility(View.INVISIBLE);
                cameraView.stop();
            }
        } catch (Exception ex) {
        }

The crash only happens sometime not everytime

cotfas, 2017-07-31:
I see it happens when I`m using my own camera permission request (by Dexter library)

If I`m calling start (even in Handler/Runnable) and without permission request, the CameraKit shows the permission request (from inside the library) and after the user approves the camera permission request, the camera appears and no crash. If I`m requesting the camera start from the Dexter camera permission request, sometimes appears the crash.