CameraKit/camerakit-android #85: Why is CameraKit only relying on Camera1 even if Camera2 is included in the library?

rt-bloombees, 2017-05-05:
From `CameraKit.java`, line 103, we can find:

`mCameraImpl = new Camera1(mCameraListener, mPreviewImpl);`

I'm getting some crashes from Android's 6.0, all related to `Camera1.java` and I was wondering if using `Camera2.java` for API 21 and above would avoid them. 

Why is CameraKit always instantiating Camera1 for all Android versions?

xp-vit, 2017-05-05:
If I understand correctly, Camera2 just isn't implemented and is there just for future support. Currently CameraView class just creates Camera1 independently which device it's running on:
https://github.com/gogopop/CameraKit-Android/blob/master/camerakit/src/main/java/com/flurgle/camerakit/CameraView.java#L103

kristjanmik, 2017-07-24:
Is there any timeline to implement the Camera 2 class? Currently the name of the project is confusing since it indicates that Camera2 is being used under the hood

developer--, 2017-10-30:
would be great

austinkettner, 2017-10-30:
@developer-- @kristjanmik @rt-bloombees We plan to include full Camera2 support, in a very robust way by v1.0.0 of the library. We are currently about to be releasing 0.12.0 and expect v1.0.0 by the first of the year.

The inclusion of Camera2 is due to long term needs and some short term decisions even though not in use. However, we do have full plans to utilize this!

I have slotted this ticket into the v1.0.0 milestone to indicate it's intended release window.

emersoncloud, 2018-07-30:
@doyee we currently have camera2 support on the latest release 1.0.0 beta1! Check out implementation details at #318

closed 2018-07-30 by emersoncloud