CameraKit/camerakit-android #87: Crash when focus

Ztiany, 2017-05-12:
A continuous click view triggers focus that may cause the program to crash

CameraView :

```
    <com.flurgle.camerakit.CameraView
        android:id="@+id/camera_view"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        app:ckCropOutput="false"
        app:ckFacing="back"
        app:ckFlash="off"
        app:ckFocus="tapWithMarker"
        app:ckJpegQuality="100"
        app:ckMethod="standard"/>
```

Log : 

```
java.lang.RuntimeException: setParameters failed
                                                                               at android.hardware.Camera.native_setParameters(Native Method)
                                                                               at android.hardware.Camera.setParameters(Camera.java:2206)
                                                                               at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:533)
                                                                               at android.view.View.dispatchTouchEvent(View.java:9300)
                                                                               at com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:130)
                                                                               at android.view.View.dispatchTouchEvent(View.java:9300)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2548)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2241)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
                                                                               at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
                                                                               at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2405)
                                                                               at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1737)
                                                                               at android.app.Activity.dispatchTouchEvent(Activity.java:2835)
                                                                               at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                                               at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2366)
                                                                               at android.view.View.dispatchPointerEvent(View.java:9526)
                                                                               at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4283)
                                                                               at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4149)
                                                                               at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3690)
                                                                               at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3743)
                                                                               at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3709)
                                                                               at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3835)
                                                                               at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3717)
                                                                               at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3892)
                                                                               at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3690)
                                                                               at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3743)
                                                                               at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3709)
                                                                               at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3717)
                                                                               at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3690)
                                                                               at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5975)
                                                                               at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5949)
                                                                               at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5910)
                                                                               at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6078)
                                                                               at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                                               at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                               at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                               at android.os.Looper.loop(Looper.java:135)
                                                                               at android.app.ActivityThread.main(ActivityThread.java:5458)
                                                                               at java.lang.reflect.Method.invoke(Native Method)
                                                                               at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
                                                                               at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)
```

serghov, 2017-08-31:
I get the same error.

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc