CameraKit/camerakit-android #89: RuntimeException startPreview failed

smoan22, 2017-05-13:
This error occurs randomly and that's more annoying. Whenever i open the activity in which i am setting up camerakit layout i get Fatal Exception: java.lang.RuntimeException startPreview failed. Full stacktrace is:

`Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.flurgle.camerakit.Camera1.start(Camera1.java:87)
       at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
       at java.lang.Thread.run(Thread.java:818)
`
I am facing 4-5 different errors and i am think of switching cause i am not able to find any answers so hopefully i'll be able to get answers here.

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc

berviantoleo, 2018-05-25:
I still face this problem in version 0.13.1, I get this problem in LG Device

ArsalRaza, 2018-09-22:
I am also facing this issue in version 0.13.1 .

emersoncloud, 2018-09-24:
@ArsalRaza Which version of Android are you using? Can you post the code you are using to create the `CameraKitView`? Both the XML and java activity. Thanks!

FreelancerNI, 2019-05-15:
> @ArsalRaza Which version of Android are you using? Can you post the code you are using to create the `CameraKitView`? Both the XML and java activity. Thanks!

I am also facing this issue in version 0.13.1  on recording video

Device name : LG g4 stylus
Android Version : 6.0

@emersoncloud