CameraKit/camerakit-android #95: Image Orientation Problem

Imadk0201, 2017-05-22:
For some reason, on Samsung Galaxy Devices, whenever an image is captured from my camera, the conversion of it to a byte array and then later a bitmap disorients the image entirely by rotating it 90 degrees. This issue was not noted on any other device (Also tested on Nexus devices), but for Galaxies, this issue has been unresolved. Some images of the preview versus displayed bitmap are presented. Is there a solution to this?

[image]
[image]

Dewinster, 2017-05-22:
This could be the same issue as #36 .
Please take a look over there.

hablema, 2017-05-25:
Hi,

camerakit:ckMethod="still" is a nice way to solve the problem as the preview is returned but is there any other way so that i can use the flash which works only in standard mode when image is taken from camera ?

hablema, 2017-05-25:
@dwillmc  :  i solved my issue using this to find the offset i need to set for the orientation. It would be awesome if someone can integrate this into the standard method of image capture so it works same in all devices especially Samsung devices. Have tested in few devices (12) that i have and it works perfectly fine. Do check it out.

http://grepcode.com/file/repository.grepcode.com/java/ext/com.google.android/android-apps/4.1.1_r1/com/android/camera/Exif.java

Dewinster, 2017-05-30:
Hmm, every picture contains Exif data with an rotation tag. 
I think that the best practice would be to read the rotation when placing the image into the view.

I noticed that the latest version of Picasso don't handle this correctly but Glide does. 
May using Glide to place image data into a view would be a solution.

Johnson145, 2017-06-14:
For anybody else having these issues:
I finally implemented a version that seems to be able handling any device correctly. I'm following the same approach that @hablema already mentioned, but I'm assuming a few things:
- the associated activity is using portrait mode only
( images taken in landscape will still be rotated correctly though)
- the captured image should be saved on disk

The image itself will not be rotated, but the exif tag will be corrected such that you can open and display it properly. Of course, you could replace the exif tag with an actual rotation, too.

Checkout my [code example](https://gist.github.com/Johnson145/0e913541ce5c40147039ddb675e7c472).

KingOfSu, 2017-07-03:
@Imadk0201 Excuse me,  Can you tell me how do you normally display the preview picture ？i have this issues .

closed 2018-02-03 by austinkettner