CameraKit/camerakit-android #96: Camera is on tablet stretched out

joeykieboom, 2017-05-24:
this is the tablet version, its stretched out...
[image]

this is the phone version, it looks pretty good..
[image]

I have no idea what makes it do this, is there anybody who can help me?

Mattimon, 2017-08-15:
I've got the same problem. I use different tablets for testing and on both android 6.0 tablets i got this issue. the one with android 4.2.2 is working pretty well. could it be a problem with the camera2 api?
`<com.flurgle.camerakit.CameraView xmlns:camerakit="http://schemas.android.com/apk/res-auto"
                android:id="@+id/camera"
                android:layout_width="match_parent"
                android:layout_height="wrap_content"
                camerakit:ckFacing="front"
                android:adjustViewBounds="true" />`

tartacutza, 2017-10-16:
I face something similar when I open my camera activity (which is android:screenOrientation="sensorLandscape") and the phone is vertical when this starts. If I rotate my phone to landscape before or simply close and reopen the screen, the camera if refreshed and the stretching disappears.

closed 2018-02-03 by austinkettner