CameraKit/camerakit-android #98: Tapping capture button rapidly causes RuntimeException

noln, 2017-05-26:
When tapping rapidly on shutter button in demo app, thereby triggering the `captureImage()` method, a fatal RuntimeException is triggered because the camera is already busy with the first request.

```
05-26 09:28:55.217 23860-23860/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.flurgle.camerakit.demo, PID: 23860
                                                   java.lang.RuntimeException: takePicture failed
                                                       at android.hardware.Camera.native_takePicture(Native Method)
                                                       at android.hardware.Camera.takePicture(Camera.java:1434)
                                                       at com.flurgle.camerakit.Camera1.captureImage(Camera1.java:211)
                                                       at com.flurgle.camerakit.CameraView.captureImage(CameraView.java:318)
                                                       at com.flurgle.camerakit.demo.MainActivity.capturePhoto(MainActivity.java:124)
                                                       at com.flurgle.camerakit.demo.MainActivity_ViewBinding$3.doClick(MainActivity_ViewBinding.java:72)
                                                       at butterknife.internal.DebouncingOnClickListener.onClick(DebouncingOnClickListener.java:22)
                                                       at android.view.View.performClick(View.java:5637)
                                                       at android.view.View$PerformClick.run(View.java:22429)
                                                       at android.os.Handler.handleCallback(Handler.java:751)
                                                       at android.os.Handler.dispatchMessage(Handler.java:95)
                                                       at android.os.Looper.loop(Looper.java:154)
                                                       at android.app.ActivityThread.main(ActivityThread.java:6121)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

JakeWoki, 2017-05-31:
+1
Device: Honor SCL-AL00
OS: EMUI3.1.1(Android 5.1.1)

xxxifan, 2017-06-12:
Better to avoid quick tap in 1s

Johnson145, 2017-06-12:
> Better to avoid quick tap in 1s
Sure, this would be a quick fix for most situations, but nothing guarantees me, that all devices will handle the request during 1s. It would be better to block further requests until a callback of the first one has been catched. Maybe, it's already enough to use the existing ones: onPictureTaken/onVideoTaken?

noln, 2017-06-12:
The callback mechanism works. The fix is in #99, ready to merge. It has been for nearly three weeks now.

Feel free to validate it on your devices and if it works for you too, comment to try and nudge the repo owner to accept some PRs. There's a load that have been waiting for a while now to be merged or rejected. 

Beginning to think this project has been abandoned..

Johnson145, 2017-06-15:
Btw, the same problem arises for the button that toggles the camera view. Tapping it very fast results in an app crash, too. I think this is not yet fixed in @noln 's changes either, is it?
For now, I'm disabling all buttons after using them manually. For the case of toggling the camera, they can be enabled again on the onCameraOpened event.

noln, 2017-06-15:
Good spot @Johnson145. Hadn't tried the same with the other buttons. Will get a fix in if I have a chance this week, though seeing as the PRs aren't getting processed I might just keep JitPacking my own fork for now.

closed 2017-06-16 by dwillmc

JakeWoki, 2017-07-11:
``` Java
Process: com.flurgle.camerakit.demo, PID: 18496
java.util.ConcurrentModificationException
	at java.util.LinkedHashMap$LinkedHashIterator.nextEntry(LinkedHashMap.java:411)
	at java.util.LinkedHashMap$KeyIterator.next(LinkedHashMap.java:422)
	at android.hardware.Camera$Parameters.flatten(Camera.java:3259)
	at android.hardware.Camera.setParameters(Camera.java:2586)
	at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:446)
	at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:375)
	at com.flurgle.camerakit.Camera1.start(Camera1.java:96)
	at com.flurgle.camerakit.CameraView$3.run(CameraView.java:301)
	at android.os.Handler.handleCallback(Handler.java:761)
	at android.os.Handler.dispatchMessage(Handler.java:98)
	at android.os.Looper.loop(Looper.java:156)
	at android.os.HandlerThread.run(HandlerThread.java:61)
```
Device: Honor 8 Lite
OS: EMUI5.0(Android 7.0)

PR #99 (2017-06-16): Fixed Android Studio preview rendering issue, and 'captureImage' RuntimeException
Changes as per commit comments.
Fixes #76 and Fixes #98.