CameraKit/camerakit-android #100: Error: Failed to resolve: com.flurgle:camerakit:0.9.17

vzool, 2017-05-29:
Hi,

I'm facing an error to install your library via Gradle.

Here is my dependancies:

```Gradle
buildscript {
    repositories {
        mavenCentral()
        maven { url 'https://maven.fabric.io/public' }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:2.2.3'
        classpath 'io.fabric.tools:gradle:1.+'
    }
}
apply plugin: 'com.android.application'
apply plugin: 'io.fabric'

repositories {
    maven { url 'https://maven.fabric.io/public' }
}

dependencies {
    compile('com.android.support:multidex:1.0.0')
    compile fileTree(dir: 'libs', include: '*.jar')
    compile 'com.android.support:appcompat-v7:19.1.0'
    compile 'com.google.android.gms:play-services-gcm:8.1.0'
    compile('com.crashlytics.sdk.android:crashlytics:2.6.8@aar') {
        transitive = true;
    }
    compile 'com.google.android.gms:play-services-appindexing:8.1.0'
    compile 'com.flurgle:camerakit:0.9.17'
}

android {
    compileSdkVersion 17
    buildToolsVersion "23.0.2"

    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt')
        }
    }

    dexOptions {
//        preDexLibraries = false
    }

    defaultConfig{
        // Enabling multidex support.
        multiDexEnabled true
    }

    sourceSets {
        main {
            manifest.srcFile 'AndroidManifest.xml'
            java.srcDirs = ['src']
            resources.srcDirs = ['src']
            aidl.srcDirs = ['src']
            renderscript.srcDirs = ['src']
            res.srcDirs = ['res']
            assets.srcDirs = ['assets']
        }

        // Move the tests to tests/java, tests/res, etc...
        instrumentTest.setRoot('tests')

        // Move the build types to build-types/<type>
        // For instance, build-types/debug/java, build-types/debug/AndroidManifest.xml, ...
        // This moves them out of them default location under src/<type>/... which would
        // conflict with src/ being used by the main source set.
        // Adding new build types or product flavors should be accompanied
        // by a similar customization.
        debug.setRoot('build-types/debug')
        release.setRoot('build-types/release')
    }

}

configurations {
    all*.exclude group: 'com.android.support', module: 'support-v4'
}
```

What is wrong?

Thanks

eduardo4jesus, 2017-06-02:
What is the error message?

I am using:

```
    compile 'com.flurgle:camerakit:0.9.18'
```

dwillmc, 2017-10-11:
Please update your gradle to
```
compile 'com.wonderkiln:camerakit:0.11.1'
```

bajajra30, 2018-02-03:
I made an app in October which had flurgle instead of wonderkiln.
how to keep using that??
please help....

austinkettner, 2018-02-03:
Hey @bajajra30 for all versions now you need to use `wonderkiln` instead of `flurgle`. For v1.0.0 and on it will be a different reference as well

austinkettner, 2018-02-28:
Will look into this and republish a 0.9.18 if needed today! Will comment later today @himangipatel

bajajra30, 2018-02-28:
Hey I have a saved copy of older version. Juat save it as a library and
import to get going.

austinkettner, 2018-02-28:
@himangipatel Just use com.flurgle:camerakit for 0.9.x releases and all should work out well!

dwillmc, 2018-02-28:
The old flurgle group-name releases seem to have been taken of jcenter so they won't resolve automatically.

Can you change

```
repositories {
    maven { url 'https://maven.fabric.io/public' }
}
```

to

```
repositories {
    maven { url 'https://maven.fabric.io/public' }
    maven { url 'https://dl.bintray.com/flurgle/maven' }
}
```

himangipatel, 2018-03-01:
@austinkettner  Still I am not able to sync my com.flurgle:camerakit:0.9.18

I have also tried  below code :

repositories {
    maven { url 'https://maven.fabric.io/public' }
}
to

repositories {
    maven { url 'https://maven.fabric.io/public' }
    maven { url 'https://dl.bintray.com/flurgle/maven' }
}

himangipatel, 2018-03-01:
**@bajajra30  can you please provide me your copy version ?**

dwillmc, 2018-03-01:
@himangipatel the version you're looking for is still hosted here:

https://dl.bintray.com/flurgle/maven/com/flurgle/camerakit/0.9.18/

You should be able to download `camerakit-0.9.18.aar` and use it directly

himangipatel, 2018-03-01:
@dwillmc Yes Thank you so much

shaktisinh, 2018-05-21:
Caused by: android.view.InflateException: Binary XML file line #14: Binary XML file line #14: Error inflating class com.wonderkiln.camerakit.CameraView

implementation 'com.wonderkiln:camerakit:0.13.1'

 maven { url "https://jitpack.io" }

also tried with

 maven { url "https://jitpack.io" }
 maven { url 'https://dl.bintray.com/flurgle/maven' }

emersoncloud, 2018-05-24:
@shaktisinh if you still want to use CameraKit version 0.13.1 that implementation line is correct. Check out this repository if you want to see a simple working example of CameraKit. 

https://github.com/emersoncloud/camerakit-crash/blob/master/app/build.gradle

The `master` branch uses 0.13.1 and the `camerakit-1.0.0` branch uses CameraKit version 1.0.0.

The team has put a lot of effort into 1.0.0 and fixes numerous issues in 0.13.1. The office release is set for the end of the month but feel free to give it a shot now! You can find more information here: #318.

rajiraji123, 2021-05-05:
> As per you said I have changed my lib to com.wonderkiln:camerakit:0.9.18 from com.flurgle:camerakit:0.9.18
> But still there is error
> Failed to resolve: com.wonderkiln:camerakit:0.9.18
> 
> I don't want to upgrade to latest version. I want to keep my project to 0.9.18 version but it is not able to sync
> 
> Please help

hi @himangipatel how you resolved this issue? I am getting the same issue now.

imilakovic, 2021-05-13:
This is the output from CircleCI:
```
> Task :app:androidDependencies FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:androidDependencies'.
> Could not resolve all artifacts for configuration ':app:myFlavorDebugCompileClasspath'.
   > Could not resolve com.flurgle:camerakit:0.9.18.
     Required by:
         project :app
      > Could not resolve com.flurgle:camerakit:0.9.18.
         > Could not get resource 'https://dl.bintray.com/flurgle/maven/com/flurgle/camerakit/0.9.18/camerakit-0.9.18.pom'.
            > Could not GET 'https://dl.bintray.com/flurgle/maven/com/flurgle/camerakit/0.9.18/camerakit-0.9.18.pom'. Received status code 403 from server: Forbidden
```

Basically, CameraKit dependency cannot be downloaded because the GET request returns `403 Forbidden`.

xamantra, 2021-06-16:
We are having this issue currently.

xamantra, 2021-06-16:
Try to visit this url: https://dl.bintray.com/fotoapparat/fotoapparat/com/flurgle/camerakit/0.9.17/camerakit-0.9.17.pom

Both .17 and .18 cannot be downloaded.

xamantra, 2021-06-16:
An update from our other devs in the company:

> I didn't get this issue last friday and was able to build and launch on emulator though. Maybe the issue comes and goes