CameraKit/camerakit-android #102: Fail to connect to camera service

gaelthas, 2017-05-31:
java.lang.RuntimeException: Fail to connect to camera service
at android.hardware.Camera.<init>(Camera.java:552)
at android.hardware.Camera.open(Camera.java:376)
at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:314)
at com.flurgle.camerakit.Camera1.start(Camera1.java:85)
at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
at java.lang.Thread.run(Thread.java:818)

m-rahimy, 2017-06-29:
This happens a lot when there is a problem connecting to camera:

FATAL EXCEPTION: Thread-6
Process: ir.mrkgrahimy.camerarecordwithlibraries, PID: 3759
java.lang.RuntimeException: Fail to connect to camera service
at android.hardware.Camera.<init>(Camera.java:496)
at android.hardware.Camera.open(Camera.java:345)
at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:314)
at com.flurgle.camerakit.Camera1.start(Camera1.java:85)
at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
at java.lang.Thread.run(Thread.java:761)

GenyMotion tablet 7.1.0 emulator (either connected to dummycamera or the laptop's camera)

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc

jlmcdonnell, 2017-12-08:
I am seeing this in 0.12.1 in the Android emulator (API 26). It is fine after restarting the emulator. Is there a way to stop this error from crashing the app?

eric-vader, 2018-06-18:
Yea I am experiencing this problem:
`    java.lang.RuntimeException: Fail to connect to camera service
        at android.hardware.Camera.<init>(Camera.java:520)
        at android.hardware.Camera.open(Camera.java:361)
        at com.wonderkiln.camerakit.Camera1.openCamera(Camera1.java:667)
        at com.wonderkiln.camerakit.Camera1.start(Camera1.java:124)
        at com.wonderkiln.camerakit.CameraView$2.run(CameraView.java:266)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:211)
        at android.os.HandlerThread.run(HandlerThread.java:61)`

emersoncloud, 2018-06-18:
Hey @eric-vader thanks for using CameraKit! All of our attentions is currently on releasing our v1.0.0 which has a lot of stability improvements and enhancements. I recommend you check that out here #318, and just using the new version will likely fix your issue. Check the PR notes for our documentation on how to get started. Thanks!