CameraKit/camerakit-android #105: crash when using still mode

xxxifan, 2017-06-04:
tested device is samsung SM-G9500, method set to still and this exception will be thrown.

`android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
                                                                       at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:7913)
                                                                       at android.view.ViewRootImpl.invalidateChildInParent(ViewRootImpl.java:1373)
                                                                       at android.view.ViewGroup.invalidateChild(ViewGroup.java:5449)
                                                                       at android.view.View.invalidateInternal(View.java:14825)
                                                                       at android.view.View.invalidate(View.java:14789)
                                                                       at android.view.View.setFlags(View.java:12684)
                                                                       at android.view.View.setVisibility(View.java:8673)
                                                                       at com.kingyun.hawkeye.camera.CaptureActivity$setupCamera$1.onPictureTaken(CaptureActivity.kt:60)
                                                                       at com.flurgle.camerakit.CameraView$CameraListenerMiddleWare.onPictureTaken(CameraView.java:397)
                                                                       at com.flurgle.camerakit.Camera1$3$1.onStillProcessed(Camera1.java:227)
                                                                       at com.flurgle.camerakit.ProcessStillTask.run(ProcessStillTask.java:47)
                                                                       at java.lang.Thread.run(Thread.java:762)`

prasad456, 2017-08-17:
I have the same issue

prasad456, 2017-08-18:
@xxxifan do you had any alternative, could you plz let me know

xxxifan, 2017-08-18:
onPictureTaken() has been called in async thread, so if you want to do something related to view you need run in main thread

closed 2018-02-03 by austinkettner