CameraKit/camerakit-android #108: java.lang.NullPointerException

nailperry-zd, 2017-06-05:
```
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.takePicture(android.hardware.Camera$ShutterCallback, android.hardware.Camera$PictureCallback, android.hardware.Camera$PictureCallback, android.hardware.Camera$PictureCallback)' on a null object reference
	at com.flurgle.camerakit.Camera1.captureImage(Camera1.java:212)
	at com.flurgle.camerakit.CameraView.captureImage(CameraView.java:323)
	at com.netease.uedc.uedcdemo.module.camera.CameraSampleActivity.capturePhoto(CameraSampleActivity.java:128)
	at com.netease.uedc.uedcdemo.module.camera.CameraSampleActivity.onClick(CameraSampleActivity.java:100)
	at android.view.View.performClick(View.java:5264)
	at android.view.View$PerformClick.run(View.java:21297)
	at android.os.Handler.handleCallback(Handler.java:743)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:150)
	at android.app.ActivityThread.main(ActivityThread.java:5546)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:792)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:682)
```

thatsVishal, 2017-08-25:
Can you check if you have Intialized the Camera. Ex: 
`CameraView camera = (CameraView) findViewById(R.id.camera_view)`
where R.id.camera_view is the id of CameraView of xml declaration.

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc