CameraKit/camerakit-android #109: Camera1 start failed runtime exception when starting video recording

alez007, 2017-06-05:
Hi, 

Currently on my samsung galaxy s4, using front camera only, the video recording fails with:
`Caused by: java.lang.RuntimeException: start failed.
                                                                                   at android.media.MediaRecorder.start(Native Method)
                                                                                   at com.flurgle.camerakit.Camera1.startVideo(Camera1.java:240)
                                                                                   at com.flurgle.camerakit.CameraView.startRecordingVideo(CameraView.java:322)`

any solutions for this ? 

Thank you

passos, 2017-09-27:
+1 same problem.

tsoumalis, 2017-12-06:
This happens for me also. Any progress on this? After some research I have found that maybe the cause of this is a difference in preview size and actual video recording size. Unfortunately I don't own a device where I can reproduce this and try to solve it but I have received multiple crash reportings from S4 and S7 Edge.

closed 2018-02-03 by austinkettner