CameraKit/camerakit-android #113: Bug !!!!!!!!!!!!

MinchinWang, 2017-06-12:
# main(1)
java.lang.RuntimeException
setParameters failed

1 android.hardware.Camera.native_setParameters(Native Method)
2 android.hardware.Camera.setParameters(Camera.java:1986)
3 com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:515)
4 android.view.View.dispatchTouchEvent(View.java:9300)
5 com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:131)
6 android.view.View.dispatchTouchEvent(View.java:9300)
7 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2548)
8 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2241)
9 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
10 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
11 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
12 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
13 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
14 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
15 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
16 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
17 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2554)
18 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2255)
19 com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2403)
20 com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1737)
21 android.app.Activity.dispatchTouchEvent(Activity.java:2820)
22 com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2364)
23 android.view.View.dispatchPointerEvent(View.java:9526)
24 android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4262)
25 android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4128)
26 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3669)
27 android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3722)
28 android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3688)
29 android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3814)
30 android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3696)
31 android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3871)
32 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3669)
33 android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3722)
34 android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3688)
35 android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3696)
36 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3669)
37 android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5954)
38 android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5928)
39 android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5889)
40 android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6057)
41 android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
42 android.os.MessageQueue.nativePollOnce(Native Method)
43 android.os.MessageQueue.next(MessageQueue.java:323)
44 android.os.Looper.loop(Looper.java:135)
45 android.app.ActivityThread.main(ActivityThread.java:5441)
46 java.lang.reflect.Method.invoke(Native Method)
47 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:738)
48 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:628)

xiuqianli1996, 2017-06-26:
I also encountered the same problem, did you solve this problem?

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc