CameraKit/camerakit-android #116: setParameters failed

mhwu1103, 2017-06-20:
After #111 , always setParameters failed.

Device: Sony Z2
OS: 6.0.1

`Process: com.flurgle.camerakit.demo, PID: 17302
java.lang.RuntimeException: Unable to resume activity {com.flurgle.camerakit.demo/com.flurgle.camerakit.demo.MainActivity}: java.lang.RuntimeException: setParameters failed
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3160)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3191)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2529)
	at android.app.ActivityThread.access$900(ActivityThread.java:154)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1391)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:5526)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
Caused by: java.lang.RuntimeException: setParameters failed
	at android.hardware.Camera.native_setParameters(Native Method)
	at android.hardware.Camera.setParameters(Camera.java:1939)
	at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:141)
	at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:433)
	at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:361)
	at com.flurgle.camerakit.Camera1.start(Camera1.java:86)
	at com.flurgle.camerakit.CameraView.start(CameraView.java:301)
	at com.flurgle.camerakit.demo.MainActivity.onResume(MainActivity.java:95)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1259)
	at android.app.Activity.performResume(Activity.java:6361)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3149)
	... 10 more`

adamwlarson, 2017-07-05:
Can you see if OnResume is being called twice in your project, mine is for some reason which is what causes this exact same problem.

mhwu1103, 2017-07-06:
@adamwlarson 
Thanks your comment, 

This report issue is using demo project, 

It's not called twice in this demo project, 

maybe is another reason.

eduardo4jesus, 2017-08-08:
I have getting `setParameters failed` every time when `Camera1.resetFocus` is called.

Johnson145, 2017-08-09:
@eduardo4jesus does #140 solve your issue?

I guess your log looks a little bit different than @mhwu1103 's, right? I'm getting almost the same log as @mhwu1103, but can't see any relation to the resetFocus.

eduardo4jesus, 2017-08-10:
Wow, Thanks a lot @Johnson145 .

#140 does solve my issue. However this solution causes it to freeze the camera for a little while sometimes. On logcat I am getting:

`I/Choreographer: Skipped 92 frames!  The application may be doing too much work on its main thread.`

In my temporary solution, I had to comment the call `params.setMeteringAreas(null);` I guess #140 would be a more correct solution, since we check if the `FOCUS_MODE_CONTINUOUS_PICTURE` is supported before setting it while still resetting the `metering areas` to `null`.

Another detail that might cause issues is the `!=` for comparing `Strings` on this method. I replaced it by `equals` instead, resulting in the code bellow:
```
if (!params.getFocusMode().equals(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE)) {
                        List<String> modes = params.getSupportedFocusModes();
                        if (modes.contains(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE)) {
                            params.setFocusMode(Camera.Parameters.FOCUS_MODE_CONTINUOUS_PICTURE);
                        }
                        params.setFocusAreas(null);
                        params.setMeteringAreas(null);
                        camera.setParameters(params);
                    }
```

adamwlarson, 2017-08-10:
I had something similar that I ended up doing this change to fix https://github.com/adamwlarson/CameraKit-Android/commit/3e59b6dbb6dae0573962626e70ed9ca918b99252

setParameters was failing because of the calculation of capture size

PR #267 (2017-12-07): Release 0.13.0
Some goals before merging...
- [ ] Reduce Code Climate error count
- [ ] Update README
- [ ] Update Tickets Resolved by Merge in this area so they close on merge
- [ ] Reference new Google Play app

Resolve #248 
Resolve #233 
Resolve #154 
Resolve #116
closed 2017-12-07 by dwillmc