CameraKit/camerakit-android #117: Camera crashing when closing and opening camera frequently.

sw-tt-rahulshah, 2017-06-22:
Application is always crashing when I frequently foreground to background and vice versa when camera screen in showing. Please help.

```
E/AndroidRuntime: FATAL EXCEPTION: main
                                                                     Process: com.googlefbintegration, PID: 624
                                                                     java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                         at android.hardware.Camera.setPreviewTexture(Native Method)
                                                                         at com.flurgle.camerakit.Camera1.setupPreview(Camera1.java:330)
                                                                         at com.flurgle.camerakit.Camera1.access$100(Camera1.java:29)
                                                                         at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:70)
                                                                         at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
                                                                         at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
                                                                         at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
                                                                         at android.view.TextureView.getHardwareLayer(TextureView.java:368)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15157)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.support.v4.widget.DrawerLayout.drawChild(DrawerLayout.java:1379)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15175)
                                                                         at android.view.View.draw(View.java:15954)
                                                                         at android.view.ViewGroup.drawChild(ViewGroup.java:3609)
                                                                         at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3399)
                                                                         at android.view.View.draw(View.java:16187)
                                                                         at com.android.internal.policy.PhoneWindow$DecorView.draw(PhoneWindow.java:2690)
                                                                         at android.view.View.updateDisplayListIfDirty(View.java:15180)
                                                                         at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:281)
                                                                         at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:287)
                                                                         at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:322)
                                                                         at android.view.ViewRootImpl.draw(ViewRootImpl.java:2615)
                                                                         at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2434)
                                                                         at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2067)
                                                                         at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1107)
                                                                         at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:6013)
                                                                         at android.view.Choreographer$CallbackRecord.run(Choreographer.java:858)
                                                                         at android.view.Choreographer.doCallbacks(Choreographer.java:670)
                                                                         at android.view.Choreographer.doFrame(Choreographer.java:606)
                                                                         at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
                                                                         at android.os.Handler.handleCallback(Handler.java:739)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                     	at android.os.Loope
```

JakeWoki, 2017-07-11:
+1
Device: Honor 8 Lite
OS: EMUI5.0(Android 7.0)

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc