CameraKit/camerakit-android #119: Fatal Exception: java.lang.NullPointerException

TheLester, 2017-06-26:
0. Crashed: main: 0 0 0x0000000000000000
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:358)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
       at android.view.TextureView.getHardwareLayer(TextureView.java:389)
       at android.view.TextureView.draw(TextureView.java:338)
       at android.view.View.updateDisplayListIfDirty(View.java:17291)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18316)
       at android.view.View.updateDisplayListIfDirty(View.java:17291)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17286)
       at android.view.View.draw(View.java:18075)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18316)
       at com.android.internal.policy.DecorView.draw(DecorView.java:850)
       at android.view.View.updateDisplayListIfDirty(View.java:17291)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:666)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:672)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:780)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:3103)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2899)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2493)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1493)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7040)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:927)
       at android.view.Choreographer.doCallbacks(Choreographer.java:702)
       at android.view.Choreographer.doFrame(Choreographer.java:638)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6688)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)

--

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.bj.a()' on a null object reference
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:358)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)

sk8darr, 2017-10-09:
Hi, did you find a way to fix this error?

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc