CameraKit/camerakit-android #130: Camera is being used after Camera.release() was called

JakeWoki, 2017-07-11:
``` Java
Process: com.flurgle.camerakit.demo, PID: 19753
java.lang.RuntimeException: Camera is being used after Camera.release() was called
	at android.hardware.Camera.startPreview(Native Method)
	at com.flurgle.camerakit.Camera1$2.onPictureTaken(Camera1.java:233)
	at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1185)
	at android.os.Handler.dispatchMessage(Handler.java:105)
	at android.os.Looper.loop(Looper.java:156)
	at android.os.HandlerThread.run(HandlerThread.java:61)
```
Device: Honor 8 Lite
OS: EMUI5.0(Android 7.0)

eduardo4jesus, 2017-08-08:
I am having this same problem, but it's happening in `onTouch` event

```
                                                                    java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                        at android.hardware.Camera.native_getParameters(Native Method)
                                                                        at android.hardware.Camera.getParameters(Camera.java:1952)
                                                                        at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:581)
                                                                        at android.view.View.dispatchTouchEvent(View.java:10771)
                                                                        at com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:174)
                                                                        at android.view.View.dispatchTouchEvent(View.java:10771)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2859)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2535)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2865)
                                                                        at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2550)
                                                                        at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:583)
                                                                        at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1884)
                                                                        at android.app.Activity.dispatchTouchEvent(Activity.java:3240)
                                                                        at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                                                                        at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:545)
                                                                        at android.view.View.dispatchPointerEvent(View.java:11004)
                                                                        at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5124)
                                                                        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4976)
                                                                        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4501)
                                                                        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4554)
[... cut ...]
                                                                        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                                        at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                        at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                        at android.os.Looper.loop(Looper.java:136)
```

eduardo4jesus, 2017-08-10:
I can reproduce my problem really easy, basically I just rotate my screen, and as soon it starts responding, I rotate it back.

This is my stack:

```
E/AndroidRuntime: FATAL EXCEPTION: main
                                                                      Process: com.company.app, PID: 30186
                                                                      java.lang.RuntimeException: Unable to pause activity {com.company.app/com.company.app.MainActivity}: java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                          at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4432)
                                                                          at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4391)
                                                                          at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:5167)
                                                                          at android.app.ActivityThread.access$1100(ActivityThread.java:219)
                                                                          at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1741)
                                                                          at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                          at android.os.Looper.loop(Looper.java:145)
                                                                          at android.app.ActivityThread.main(ActivityThread.java:6939)
                                                                          at java.lang.reflect.Method.invoke(Native Method)
                                                                          at java.lang.reflect.Method.invoke(Method.java:372)
                                                                          at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
                                                                       Caused by: java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                          at android.hardware.Camera._stopPreview(Native Method)
                                                                          at android.hardware.Camera.stopPreview(Camera.java:803)
                                                                          at com.flurgle.camerakit.Camera1.stop(Camera1.java:103)
                                                                          at com.flurgle.camerakit.CameraView.stop(CameraView.java:312)
                                                                          at com.company.app.userinterface.widgets.Camera.onPause(Camera.java:46)
                                                                          at com.company.app.userinterface.screens.CameraFragment.onPause(CameraFragment.java:95)
                                                                          at android.app.Fragment.performPause(Fragment.java:2318)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:955)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1083)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1065)
                                                                          at android.app.FragmentManagerImpl.dispatchPause(FragmentManager.java:1901)
                                                                          at android.app.Fragment.performPause(Fragment.java:2315)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:955)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1083)
                                                                          at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1065)
                                                                          at android.app.FragmentManagerImpl.dispatchPause(FragmentManager.java:1901)
                                                                          at android.app.Activity.performPause(Activity.java:6758)
                                                                          at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1338)
                                                                          at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4418)
                                                                          at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4391) 
                                                                          at android.app.ActivityThread.handleRelaunchActivity(ActivityThread.java:5167) 
[... cut ...]
                                                                          at java.lang.reflect.Method.invoke(Native Method) 
                                                                          at java.lang.reflect.Method.invoke(Method.java:372) 
                                                                          at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404) 
                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199) 
```

tomeroto, 2017-09-17:
Have a solution for this issue?

scrobot, 2017-09-18:
Yes, solution in my last pull request)

scrobot, 2017-09-18:
https://github.com/gogopop/CameraKit-Android/pull/173

bruferrari, 2017-09-28:
any solutions on this problem?

bruferrari, 2017-09-28:
@austinkettner I saw that you assigned to you these bug, are you working on it? Can I expect a fix soon?

austinkettner, 2017-09-28:
Yes @bruferrari we will address this soon. Currently expect it sometime next week, while I know you would like it sooner, I don't want to promise something that won't happen.

closed 2017-10-06 by austinkettner

tartacutza, 2017-10-16:
Hi, this happened to me on 11.0.2 today, when rotating the device. Exactly like @eduardo4jesus mentioned.

Here is my stack
`java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                    at android.hardware.Camera.setDisplayOrientation(Native Method)
                                                                                    at com.wonderkiln.camerakit.Camera1.setDisplayAndDeviceOrientation(Camera1.java:149)
                                                                                    at com.wonderkiln.camerakit.CameraView$1.onDisplayOrDeviceOrientationChanged(CameraView.java:171)
                                                                                    at com.wonderkiln.camerakit.DisplayOrientationDetector.dispatchOnDisplayOrDeviceOrientationChanged(DisplayOrientationDetector.java:92)
                                                                                    at com.wonderkiln.camerakit.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:65)
                                                                                    at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
                                                                                    at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:830)
                                                                                    at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                                    at android.os.MessageQueue.next(MessageQueue.java:325)
                                                                                    at android.os.Looper.loop(Looper.java:142)
                                                                                    at android.app.ActivityThread.main(ActivityThread.java:6541)
                                                                                    at java.lang.reflect.Method.invoke(Native Method)
                                                                                    at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                                                                    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:767)`

tartacutza, 2017-11-06:
Hi, I see this still occurred in 0.12.0, any updates on it?

MrDizzle, 2018-05-17:
Any news on this one?

austinkettner, 2018-05-17:
This should be fully resolved in our 1.0.0 PR (#318) Check it out!

MrDizzle, 2018-05-17:
Awesome, really appreciate what you guys are doing, the Camera2 API is a horror show!

PR #188 (2017-10-06): Bug Fixes and Improvements
Constitutes a 0.11.0

Resolves #130
Resolves #153

Resolves many more, to be closed independently.