CameraKit/camerakit-android #131: NPE in Camera1.java

sonjameyer, 2017-07-11:
We are getting quite a few crashes related to camera parameters.  This is one from Crashlytics:

Fatal Exception: java.lang.NullPointerExceptionAttempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference 
  com.flurgle.camerakit.Camera1.adjustCameraParameters (Camera1.java:372)
  com.flurgle.camerakit.Camera1.access$200 (Camera1.java:30)
  com.flurgle.camerakit.Camera1$1.onSurfaceChanged (Camera1.java:75)
  com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged (PreviewImpl.java:37)
 com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged (TextureViewPreview.java:80)
 com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable (TextureViewPreview.java:26)

capezzbr, 2017-08-15:
Same crash here, latest version

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'int com.flurgle.camerakit.Size.getWidth()' on a null object reference
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:358)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
       at android.view.TextureView.getHardwareLayer(TextureView.java:389)
       at android.view.TextureView.draw(TextureView.java:338)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at com.android.internal.policy.DecorView.draw(DecorView.java:919)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:692)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:698)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:806)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:3135)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2931)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2523)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1522)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7098)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:927)
       at android.view.Choreographer.doCallbacks(Choreographer.java:702)
       at android.view.Choreographer.doFrame(Choreographer.java:638)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6682)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
```

cotfas, 2017-08-17:
Hello,

Any news about this crash? I also get this error..

Best regards

serghov, 2017-08-30:
Got the same crush on android 7.1.1

TifonT, 2017-09-06:
Same crash 
ASUS_X00GD Android 7.0

cotfas, 2017-09-06:
It seems that this library is no longer maintained :(

TifonT, 2017-09-06:
If I remove from my code `cameraView.getCaptureSize()` it not crashing

AndrewGable, 2017-09-06:
We are also seeing a similar crash with Samsung devices on OS version 7.0:
- Galaxy S7 Edge
- Galaxy S7
- Galaxy S6
- Galaxy Note5

Happens on version `0.9.17` and `0.9.18` of the library.

AndrewGable, 2017-09-06:
FYI- If anyone want's to earn a quick $500, I've posted this on Upwork to be fixed (https://www.upwork.com/jobs/~0133b12a9ecd93e6e2).

PR #179 (2017-10-05): Resolve various fatal exceptions
- Resolve #131 via `5489e34` 
- Resolve #146 via `0daf979`
- Introduction of EventListener via `6a2b521`
closed 2017-10-05 by dwillmc