CameraKit/camerakit-android #132: Crash on setParameters Camera1

sonjameyer, 2017-07-11:
We are getting quite a few crashes in a live app related to camera parameters. Another one from Crashlytics:

Fatal Exception: java.lang.RuntimeException
setParameters failed 
 android.hardware.Camera.native_setParameters (Camera.java)
 android.hardware.Camera.setParameters (Camera.java:1763)
 com.flurgle.camerakit.Camera1$5.run (Camera1.java:586)

vamzi, 2017-07-19:
It might be because of native permission glitch in the app that was mentioned in readme.md, Just dont use cameraView.start() which will crash the application after the first lauch.

hb-smeyer, 2017-07-21:
This is my code to check the permission:
@Override
    public void onStart() {
        super.onStart();
        if (ContextCompat.checkSelfPermission(this,
                Manifest.permission.CAMERA)
                != PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(this,
                    new String[]{Manifest.permission.CAMERA},
                    CAMERA_PERMISSION_REQUEST_CODE);
        }
        else {
            permissionGranted = true;
        }
    }

    @Override
    protected void onResume() {
        super.onResume();
        if (permissionGranted) {
            mCamera.start();
            enableCamera();
            mNoPermissionText.setVisibility(View.GONE);
        }
        else {
            disableCamera();
            mNoPermissionText.setVisibility(View.VISIBLE);
        }
    }

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc