CameraKit/camerakit-android #136: Crashed on start demo example

ukevgen, 2017-07-18:
`07-18 13:17:55.084 3790-3924/com.flurgle.camerakit.demo I/OpenGLRenderer: Initialized EGL, version 1.4
07-18 13:17:55.114 3790-3790/com.flurgle.camerakit.demo D/ScrollView:  onsize change changed 
07-18 13:17:55.154 3790-3790/com.flurgle.camerakit.demo W/PathParser: Points are too far apart 4.000000596046461
07-18 13:17:55.184 3790-3924/com.flurgle.camerakit.demo D/libGLESv1: DTS_GLAPI : DTS is not allowed for Package : com.flurgle.camerakit.demo
07-18 13:17:55.194 3790-3907/com.flurgle.camerakit.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                          Process: com.flurgle.camerakit.demo, PID: 3790
                                                                          java.lang.RuntimeException: setParameters failed
                                                                              at android.hardware.Camera.native_setParameters(Native Method)
                                                                              at android.hardware.Camera.setParameters(Camera.java:1984)
                                                                              at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:151)
                                                                              at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:440)
                                                                              at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:371)
                                                                              at com.flurgle.camerakit.Camera1.start(Camera1.java:96)
                                                                              at com.flurgle.camerakit.CameraView$3.run(CameraView.java:301)
                                                                              at android.os.Handler.handleCallback(Handler.java:739)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                              at android.os.Looper.loop(Looper.java:148)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)
07-18 13:17:55.264 3790-3790/com.flurgle.camerakit.demo W/art: Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
07-18 13:17:55.274 3790-3790/com.flurgle.camerakit.demo D/ViewRootImpl: MSG_RESIZED_REPORT: ci=Rect(0, 48 - 0, 0) vi=Rect(0, 48 - 0, 0) or=1
07-18 13:17:55.824 3790-3790/com.flurgle.camerakit.demo D/SensorManager: unregisterListener ::   
07-18 13:17:55.834 3790-3790/com.flurgle.camerakit.demo D/ViewRootImpl: #3 mView = null`

shaik41, 2017-07-19:
I face the same issue but it occurs very randomly. I think its got to do with the preview size.

Ali72, 2017-07-24:
this Issue come from  Camera1.class caused by:
  ` mCameraParameters.setPreviewSize(
                getPreviewResolution().getWidth(),
                getPreviewResolution().getHeight()
        ); `

because required size is not suppoerted by Camera.
Use getSupportedPictureSizes to get all available preview sizes.

To check which is maximum picture size available from camera:

`List<Size> allSizes = param.getSupportedPictureSizes();
Camera.Size size = allSizes.get(0); // get top size
for (int i = 0; i < allSizes.size(); i++) {
     if (allSizes.get(i).width > size.width)
       size = allSizes.get(i);
 }
//set max Picture Size
 params.setPictureSize(size.width, size.height);`

Johnson145, 2017-08-09:
@Ali72 I think you're mixing up the pairs (setPreviewSize, getSupportedPreviewSizes) and (setPictureSize, getSupportedPictureSizes). However, both cases are handling the supported sizes in getCaptureResolution() and getPreviewResolution(). So I'm afraid this isn't the actual issue.

Johnson145, 2017-08-09:
I guess this is the same issue as described in #116.

danielrees18, 2017-09-14:
Actually `getCaptureResolution()` and `getPreviewResolution()` are not handling supported sizes correctly. They are both getters of values set in `initResolutions()` which is where the source of this problem is occurring. 

`initResolutions()` is checking the supported `previewSizes` and `videoSizes` and then finding a common size between these two sets. Then, whatever size is found is _also_ set as the `pictureSize` when calling `mCameraParameters.setPictureSize()`. 

Some devices do not support the same `pictureSize` that was shared between `previewSize` and `videoSize`. Example, when testing this on a Samsung Galaxy S5, the method `initResolutions` sets the `previewSize` to `720x480` and then attempts to set the `pictureSize` to `720x480` as well. The result of `getSupportedPictureSizes()` does not contain the resolution `720x480` so the app crashes because `720x480` is an unsupported `pictureSize` for the Galaxy S5.

capezzbr, 2017-09-14:
@danielrees18 do you think you could make a PR to fix this behaviour since you're able to reproduce and also have a great understanding of the problem?

capezzbr, 2017-09-14:
We don't record videos in our app so it would be awesome if you open a PR and we can have a look at it 👍

danielrees18, 2017-09-18:
@capezzbr Wasn't able to get a PR to your fork. You can look at [the commit](https://github.com/danielrees18/CameraKit-Android/commit/25bfec8f2eddbfb967e280c8d6b20f3de6a85601) on my fork to see the changes that I made.

I was in recent contact with the owner of the project and he informed me that we should start seeing regular updates to this library in the coming week. Keep an eye out to see if those updates show up soon

capezzbr, 2017-09-18:
Thanks anyway @danielrees18 :D

> Keep an eye out to see if those updates show up soon

🤞

dwillmc, 2017-10-05:
`initResolutions()` is gone now and I've added `getVideoResolution` to the CameraImpl interface, so now when a phone offers a picture resolution that the media recorder can't use for videos, it won't crash.

closed 2017-10-05 by dwillmc