CameraKit/camerakit-android #141: Showing portrait image when the phone is in landscape

Tgo1014, 2017-07-28:
The images is okay in portrait, but as soons as the phone is rotate, the image zooms and the preview became in the wrong orientation.
That's kinda a critical problem.

My phone is a Samsung A5 2017.

Portrait
[image]

Landscape
[image]

commonsguy, 2017-07-29:
@Tgo1014: I was running into the same problem, on a Google Pixel. The problem seems tied to `com.flurgle:camerakit:0.9.17`. If I use `com.flurgle:camerakit:0.9.18` (the latest on JCenter), the problem goes away.

Tgo1014, 2017-07-30:
@commonsguy that indeed solved the issue. Thanks! But why in the main page the latest version is `com.flurgle:camerakit:0.9.17`?

closed 2017-07-30 by Tgo1014