CameraKit/camerakit-android #145: This library continues to be maintained?

bruferrari, 2017-08-08:
Hi, guys, I've made some tests with this lib last week and some other libraries too, but this one is the one that solve my entire problems without manage some changes in another forks.

By saying that I would like to know if this lib continues to be maintained, I've seen many issues and PR's opened for a while and that was a doubt point in order to use this lib in production.

Thanks in advance.

eduardo4jesus, 2017-08-08:
I have that same concern. I was using [larryng's fork](https://github.com/larryng/CameraKit-Android) for in the first stage of my project. He manage to solve one of the problems, but as you said, there are other Issues to be solved/merged.

As I need to move forward and get things done in my project, I am kind of forced to reinvent the wheel as well. I just decided to have my own fork and as I am facing the issues I will either look up for solutions here or, if I don't find any, solve it myself.

austinkettner, 2017-08-08:
@bruferrari @eduardo4jesus This is still in-development and something we care about. While the progress may be slower than desired, we are 100% focused on not introducing new issues. With that said, we do have long term plans to support this just might not be blazing fast.

@dwillmc Will be spending some time on this, this week, so you will see issues closed, PRs responded to and a new release.

bruferrari, 2017-08-08:
@eduardo4jesus it's something that I'm worried about, I've been using another lib to do the dirty job, but I'm also kinda reinventing the wheel and it's very bad for a production app because I can't count on people that have so much more knowledge than I. So, somehow it's a breaker for me to choose the best option in terms of maintenance.

_But we also know that Google Cameras APIs are very painful to handle and this library is awesome, please guys keep doing this **awesome** job! I'll help with PR's when something appear in my development path._

@austinkettner thanks for fast answer! It's good to know that you guys are focusing on keep this lib better, even if it means that we're haven't new features in near future.

dwillmc, 2017-08-08:
You guys can feel free to email me (email on my profile) anything specific you need for your production apps and the best solutions you've found and I'll be sure to get it in a release this week

bruferrari, 2017-08-08:
@dwillmc thanks for this awesome support guys!

eduardo4jesus, 2017-08-08:
@dwillmc, Thank you so much for taking a look at it!

matthewcarlreetz, 2017-08-08:
@dwillmc Any chance we could get pinch to zoom in the next release?

dwillmc, 2017-08-08:
@matthewcarlreetz Yes that's a core focus. I'll take a look at what everybody else has tried and get something in.

Hrishi1999, 2017-08-13:
BTW, why doesn't capturing work after .14?

mkonkel, 2017-08-23:
Hi! 
Any news about the new release?

austinkettner, 2017-10-11:
Hey everyone, we are closing this ticket as you can see the recent releases and support via our latest releases! If anyone runs into issue or questions, definitely feel free to open a new ticket though =)

closed 2017-10-11 by austinkettner