CameraKit/camerakit-android #146: ConcurrentModificationException Bug

jaksab, 2017-08-10:
Hi!
I get periodically inner bug.
What could it be? 
Version 0.9.18

08-10 14:13:58.835 2771-3877/com.infitting.app E/UncaughtException: java.util.ConcurrentModificationException
                                                                        at java.util.LinkedHashMap$LinkedHashIterator.nextEntry(LinkedHashMap.java:411)
                                                                        at java.util.LinkedHashMap$KeyIterator.next(LinkedHashMap.java:422)
                                                                        at android.hardware.Camera$Parameters.flatten(Camera.java:2490)
                                                                        at android.hardware.Camera.setParameters(Camera.java:1884)
                                                                        at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:144)
                                                                        at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:389)
                                                                        at com.flurgle.camerakit.Camera1.openCamera(Camera1.java:322)
                                                                        at com.flurgle.camerakit.Camera1.start(Camera1.java:89)
                                                                        at com.flurgle.camerakit.CameraView$3.run(CameraView.java:216)
                                                                        at java.lang.Thread.run(Thread.java:761)

capezzbr, 2017-08-15:
Same crash, latest version:

```
Fatal Exception: java.util.ConcurrentModificationException
       at java.util.LinkedHashMap$LinkedHashIterator.nextEntry(LinkedHashMap.java:411)
       at java.util.LinkedHashMap$KeyIterator.next(LinkedHashMap.java:422)
       at android.hardware.Camera$Parameters.flatten(Camera.java:2751)
       at android.hardware.Camera.setParameters(Camera.java:1940)
       at com.flurgle.camerakit.Camera1.setFlash(Camera1.java:139)
       at com.flurgle.camerakit.Camera1.adjustCameraParameters(Camera1.java:376)
       at com.flurgle.camerakit.Camera1.access$200(Camera1.java:29)
       at com.flurgle.camerakit.Camera1$1.onSurfaceChanged(Camera1.java:71)
       at com.flurgle.camerakit.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:37)
       at com.flurgle.camerakit.TextureViewPreview.dispatchSurfaceChanged(TextureViewPreview.java:84)
       at com.flurgle.camerakit.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:28)
       at android.view.TextureView.getHardwareLayer(TextureView.java:389)
       at android.view.TextureView.draw(TextureView.java:338)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.updateDisplayListIfDirty(View.java:17297)
       at android.view.View.draw(View.java:18086)
       at android.view.ViewGroup.drawChild(ViewGroup.java:3966)
       at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3752)
       at android.view.View.draw(View.java:18327)
       at com.android.internal.policy.DecorView.draw(DecorView.java:923)
       at android.view.View.updateDisplayListIfDirty(View.java:17302)
       at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:692)
       at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:698)
       at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:806)
       at android.view.ViewRootImpl.draw(ViewRootImpl.java:3135)
       at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:2931)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2523)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1522)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7098)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:927)
       at android.view.Choreographer.doCallbacks(Choreographer.java:702)
       at android.view.Choreographer.doFrame(Choreographer.java:638)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6682)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1534)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1424)
```

harryblam, 2017-08-21:
Same crash for me!

kuoliangkwong, 2017-09-14:
had the same crash too~ any solution so far?

PR #179 (2017-10-05): Resolve various fatal exceptions
- Resolve #131 via `5489e34` 
- Resolve #146 via `0daf979`
- Introduction of EventListener via `6a2b521`
closed 2017-10-05 by dwillmc