CameraKit/camerakit-android #151: Crash on focus on tap

kimihailv, 2017-08-14:
OnePlus 3, app crashes when tap to focus near screen borders:

`java.lang.RuntimeException: setParameters failed
                                                                           at android.hardware.Camera.native_setParameters(Native Method)
                                                                           at android.hardware.Camera.setParameters(Camera.java:2001)
                                                                           at com.flurgle.camerakit.Camera1$4.onTouch(Camera1.java:535)
                                                                           at android.view.View.dispatchTouchEvent(View.java:10040)
                                                                           at com.flurgle.camerakit.CameraView$2.onTouch(CameraView.java:131)
                                                                           at android.view.View.dispatchTouchEvent(View.java:10040)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2626)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2307)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2632)
                                                                           at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2321)
                                                                           at com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:416)
                                                                           at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1808)
                                                                           at android.app.Activity.dispatchTouchEvent(Activity.java:3119)
                                                                           at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:68)
                                                                           at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:378)
                                                                           at android.view.View.dispatchPointerEvent(View.java:10264)
                                                                           at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4468)
                                                                           at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4336)
                                                                           at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3883)
                                                                           at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3936)
                                                                           at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3902)
                                                                           at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4029)
                                                                           at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3910)
                                                                           at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4086)
                                                                           at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3883)
                                                                           at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3936)
                                                                           at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3902)
                                                                           at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3910)
                                                                           at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3883)
                                                                           at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6284)
                                                                           at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6258)
                                                                           at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6219)
                                                                           at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6387)
                                                                           at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                                           at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                           at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                           at android.os.Looper.loop(Looper.java:136)
                                                                           at android.app.ActivityThread.main(ActivityThread.java:6236)
                                                                           at java.lang.reflect.Method.invoke(Native Method)
                                                                           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:891)
                                                                           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:781)`

DennisAlund, 2017-08-22:
Seems related to #87 #84 #80 #61 #34 as mentioned in PR #101

dwillmc, 2017-10-11:
This should be fixed now with the latest release. Can you update your `build.gradle` to reference `com.wonderkiln:camerakit:0.11.1`. If the issue persists can you please re-open.

Thanks!

closed 2017-10-11 by dwillmc