CameraKit/camerakit-android #154: [Question] Custom File Path

vishalpatel1327, 2017-08-16:
Hello boss,
 This is great library  thanx for that

But I have one question   How to change default path for Camera and video save

Ninjaman494, 2017-08-24:
The library has no built in way to do this. What I did was manually move the file in onVideoTaken using file.renameTo(newFile), where newFile is a File object containing the new file path.

vishalpatelbacancy, 2017-11-06:
@dwillmc   for Custom path It will be great for us to add some line in code

At Camera1

//add at 267
 @Override
    void setVideoCapturePath(File path) {
        this.userGivenFile = path;
    }

// line no 779  
 private File getVideoFile() {
       
        if (!Environment.getExternalStorageState().equalsIgnoreCase(Environment.MEDIA_MOUNTED)) {
            return null;
        }
File mediaStorageDir ;
if(userGivenFile == null){
         mediaStorageDir = new File(Environment.getExternalStorageDirectory(), "Camera");
}else{
mediaStorageDir = userGivenFile;

}
        if (!mediaStorageDir.exists()) {
            if (!mediaStorageDir.mkdirs()) {
                return null;
            }
        }

        return new File(mediaStorageDir.getPath() + File.separator + "video.mp4");
    }

PR #267 (2017-12-07): Release 0.13.0
Some goals before merging...
- [ ] Reduce Code Climate error count
- [ ] Update README
- [ ] Update Tickets Resolved by Merge in this area so they close on merge
- [ ] Reference new Google Play app

Resolve #248 
Resolve #233 
Resolve #154 
Resolve #116
closed 2017-12-07 by dwillmc