CameraKit/camerakit-android #155: [Bug] Bug to record when using Camera1 API

bruferrari, 2017-08-16:
Device: Samsung Galaxy S4
Android Version: 4.4.2

Hi guys, I'm having this issue when trying to make a Samsung device record video, any ideas of what could be the problem? 

```
Fatal Exception: java.lang.IllegalStateException
       at android.media.MediaRecorder.stop(MediaRecorder.java)
       at com.flurgle.camerakit.Camera1.endVideo(SourceFile:240)
       at com.flurgle.camerakit.CameraView.stopRecordingVideo(SourceFile:326)
       at br.com.espnsf.ui.camera.CameraFragment.whenRecordButtonReleased(SourceFile:188)
       at br.com.espnsf.ui.camera.CameraFragment.access$whenRecordButtonReleased(SourceFile:38)
       at br.com.espnsf.ui.camera.CameraFragment$onCaptureVideoBtnPressed$1.onTouch(SourceFile:159)
       at android.view.View.dispatchTouchEvent(View.java:8102)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2411)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2144)
       at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2280)
       at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1615)
       at android.app.Activity.dispatchTouchEvent(Activity.java:2549)
       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(SourceFile:71)
       at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2228)
       at android.view.View.dispatchPointerEvent(View.java:8315)
       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4590)
       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4458)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4020)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4074)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4043)
       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4150)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4051)
       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4207)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4020)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4074)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4043)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4051)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4020)
       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6397)
       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6301)
       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6272)
       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6237)
       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6477)
       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
       at android.view.InputEventReceiver.nativeConsumeBatchedInputEvents(InputEventReceiver.java)
       at android.view.InputEventReceiver.consumeBatchedInputEvents(InputEventReceiver.java:176)
       at android.view.ViewRootImpl.doConsumeBatchedInput(ViewRootImpl.java:6450)
       at android.view.ViewRootImpl$ConsumeBatchedInputRunnable.run(ViewRootImpl.java:6496)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:803)
       at android.view.Choreographer.doCallbacks(Choreographer.java:603)
       at android.view.Choreographer.doFrame(Choreographer.java:571)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:789)
       at android.os.Handler.handleCallback(Handler.java:733)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:157)
       at android.app.ActivityThread.main(ActivityThread.java:5356)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:515)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1265)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1081)
       at dalvik.system.NativeStart.main(NativeStart.java)
```

ittaiOren, 2017-08-23:
I have the same issue:
Attempt to invoke virtual method 'void android.hardware.Camera.unlock()' on a null object reference

tsoumalis, 2017-12-06:
Any update on this? I am facing the same issue with 0.12 version with S7 Galaxy Edge in both front and rear camera and in S4 with front camera.

tsoumalis, 2017-12-08:
Check my issue here with a possible workaround if you want: https://github.com/wonderkiln/CameraKit-Android/issues/275

closed 2018-02-03 by austinkettner